trezor/trezor-mcu #11: The code is dual-licensed till Mon Jan 26 22:03:33 2015+0100

rfree, 2015-01-29:
Despite you rewritten and backdated the commit that changed licenses, the fact remains that up until "40efef" you published code on GPL license.

You still do use code that you released as GPL, AINAL but probably you are now required to keep publishing all modifications of this code as GPL too.

Could you please reverse this license change, and if not then fix the licenses in files that are derived work from your GPL code, as it (AFAIK) is GPL as well?

(Personally I do not support licensing, but if you start it by switching to less free Microsoft license then you should play by the rules of that game)

rfree, 2015-01-29:
Reference:
40efef commit is https://github.com/trezor/trezor-mcu/tree/40efefc5716366618accc2e4871fdb7a6d92d15f
and I backed it up as https://github.com/rfree/trezor-mcu-gpl/tree/40efefc5716366618accc2e4871fdb7a6d92d15f

Discussion with community of your users on Reddit: http://www.reddit.com/r/Bitcoin/comments/2u1wea/trezor_code_no_longer_lgplv3_but_now_more/?sort=top

markburns, 2015-01-30:
Looks like they changed their minds: http://satoshilabs.com/news/2015-01-30-trezor-software-license/

Good effort

prusnak, 2015-01-30:
^_^

closed 2015-01-30 by prusnak