trezor/trezor-mcu #15: 1.3.1 installed on production mytrezor.com, 1.2.1 still appearing on test.mytrezor.com

lichtamberg, 2015-02-20:
Hi,

Anybody has the same problems?

I installed 1.3.1 on the production backend, looks fine, also in the advanced tab I can see 1.3.1 (also after dis/reconnecting).

But when I switch to test.mytrezor.com, I still get the warning that 1.2.1 is installed and I should upgrade?

I cleared my cache and dis/reconnected.

lichtamberg, 2015-02-20:
Solved: Just forget the device and reconnect it - then it works fine again.

Saw this post on reddit and it worked: http://www.reddit.com/r/TREZOR/comments/2wdhhh/installed_301_but_mytrezor_says_i_still_have_30/copvsjt

Anyway, it would be good if this gets fixed...

slush0, 2015-02-20:
I though we've fixed it yesterday, but obviously it still appears for somebody. We'll dig into it more. Thanks for report!

lichtamberg, 2015-02-20:
Thanks for the quick reply. If you need some details/logs, please tell me.

slush0, 2015-02-20:
Found it. Yesterday I made a bugfix release on production mytrezor, but I forgot to update test.mytrezor.com as well. It's fixed now.

closed 2015-02-20 by slush0