trezor/trezor-mcu #19: Versioning of submodule

ELM4Ever, 2015-03-20:
I appreciate all the efforts put by contributors to improve trezor-mcu.  However, it would be a good idea to maintain a versioning of the trezor-mcu repo, in synch with a versioning of the submodules, like trezor-crypto, trezor-qrenc and trezor-common.  At the moment, because of recent changes in the submodules, it is impossible for the community to independently verify the signature of release 1.3.1.

slush0, 2015-03-20:
Firmware reports git commit hash from which it was built in Feature message. By cloning this repo, submodules are in the exact version as it was built so it is possible to review the code of specific version.

ELM4Ever, 2015-03-20:
Ok - I got it, my apologies.  I forgot to link the submodules with trezor-mcu.  Using git checkout f2f50aa, and git submodule upgrade commands, for rev 1.3.1, I was able to retrieve the submodules code at that time.  Closing.

closed 2015-03-20 by ELM4Ever