trezor/trezor-mcu #20: gcc-arm-none-eabi apt-get install

ELM4Ever, 2015-03-20:
Running ./firmware-docker-build.sh currently outputs the following error:

E: Version '4.9.3.2014q4-0trusty12' for 'gcc-arm-none-eabi' was not found

(Linux Ubuntu 14.04 LTS)

prusnak, 2015-03-21:
Sadly, PPA removes old packages when there is an update pushed to the repo. We'll have to deal with the issue somehow, but I can't think of any better way than copying packages to our own repo.  :-(

jhoenicke, 2015-04-29:
The last time I needed the old compilers (for verifying the bootloader -- and yes I could reproduce the exact image), I downloaded the binary linux package from 
 https://launchpad.net/gcc-arm-embedded/+download

Of course, there is no guarantee that they will be always available from there.

closed 2015-08-03 by prusnak

jcalfee, 2017-06-09:
I was able to build v1.4.2 (fingerprint matches) using ./firmware-docker-build.sh after upgrading the docker from 16.04 to 16.10:

`FROM ubuntu:16.10`

Just wish I knew how to load it on the device now.. lol