trezor/trezor-mcu #24: v1.3.1 or previous versions fingerprints not match

anduck, 2015-04-02:
Building trezor-mcu as instructed gives different fingerprints than what's listed on https://mytrezor.com/data/firmware/releases.json
The json says v1.3.1 fingerprint is 8030e257fc4c75a8f4a0325f8ea37428dd8fc68a5f9ec5f8e2d1a0de328860cc 
and my local build gives 7f55513a8d5ab1ebdd125a123c96cc230ed976993943007d7f2b31e23c05a630

prusnak, 2015-04-03:
Have you used the Dockerfile checkouted from v1.3.1 git release tag?

anduck, 2015-04-03:
Yes. It produces different fingerprint.

slush0, 2015-04-03:
I confirm, my local build now produces 7f555... as well. When I did local builds in time of firmware release, it made correct fingerprint, so obviously something changed in underlying OS (docker does "apt-get update" on fresh installation).

anduck, 2015-04-03:
Okay. Is there a way to produce the v1.3.1 fingerprint that matches with releases.json one?

prusnak, 2015-04-03:
I guess that's not easy without the time machine.

anduck, 2015-04-03:
Then should you guys make a firmware update with a signature that can be verified?

slush0, 2015-04-03:
There'll be 1.3.3 soon, so I think it won't make sense.

In the meantime we'll think about how to improve our process so this won't happen next time.

Pigeons, 2015-04-08:
well for starters have the docker config not "apt-get upgrade"

prusnak, 2015-04-09:
That does not help, because Docker folks can push update via modifying the ubuntu:14.04 tag.

It would make sense to use particular revhash instead of 14.04 tag and don't call the upgrade.

jhoenicke, 2015-04-30:
The problem is again arm-none-eabi-gcc.  Installing gcc-arm-none-eabi-4_9-2014q4 from https://launchpad.net/gcc-arm-embedded/+download  I could reproduce the 8030e257fc... hash.

closed 2015-08-04 by prusnak