trezor/trezor-mcu #25: Bootloader larger than 32KiB

kensenjiha, 2015-04-11:
Trying to build the bootloader, generated bootloader.bin is 39kB. gcc-arm-none-eabi 4.9.2.
seems like the trezor-crypto part is too large.

jhoenicke, 2015-04-29:
For me the bootloader is 24k.  Try again by calling `make clean && make` in the bootloader directory.

Firmware and bootloader compile trezor-crypto with different flags: bootloader enables `USE_PRECOMPUTED_CP=0` to save space.  However, there is no automatic clean before compiling the bootloader, so if you compiled firmware before you get the large crypto library.

kensenjiha, 2015-04-30:
hmm, I accidentally dropped `-Wl,--gc-sections` while changing `LD` from `gcc` to `ld`. now the size is ~24kB. 
Thanks~

closed 2015-04-30 by kensenjiha