trezor/trezor-mcu #33: Double size of the font for some information shown on display

ondra-novak, 2015-06-30:
It would be nice to have some crucial information shown on the display by a larger (double sized) font. For example during the recovery phase, the number of the word should be larger for easier read. 

I understand, that the firmware can have just a single font for limited memory space - to scale font using double pixels for each axis is also an acceptable solution.

Seed words should be also larger or at least shown by capitals

jhoenicke, 2015-07-01:
I did larger numbers (using the pinmatrix bitmaps) and double-sized letters (scaled) in my [letter-by-letter recovery](https://github.com/jhoenicke/trezor-mcu/tree/letterrecovery) experimental patch.  For recovery this is very helpful as you have to alternate between computer and TREZOR display.

I think a larger high-resolution font should easily fit on the TREZOR (it still has enough free read-only memory), but the font routines have to be rewritten to support fonts that are more than 8 pixels high.

prusnak, 2015-11-18:
Implemented in 50c8811af970754a47452efec1b4df558c8ba491, char zooming taken from https://github.com/jhoenicke/trezor-mcu/commit/e7b12b3bf7c0640c5f9543c9247a71193b8e1936

closed 2015-11-18 by prusnak