trezor/trezor-mcu #37: Unable to send transaction from Trezor with 2.3.2.2

Dehumanizer77, 2015-07-09:
Recently I have upgraded from Electrum 2.0 which worked fine to 2.3.2.2 and since then, I am unable to send transactions. When I submit a transaction, it asks for a confirmation and then the error message pops up, saying `Got <class 'trezorlib.messages_pb2.PublicKey'>, expected (<class 'trezorlib.messages_pb2.Success'>,)`
The old 2.0 version still works fine though. 

Note that I have the same problem with regular Electrum and I've submited the issue here: https://github.com/spesmilo/electrum/issues/1353
Since I'm not sure if it will get into electrum-ltc code after it will be fixed there, I'm also submiting the issue here. Feel free to close it if you feel it's a duplicate.

prusnak, 2015-07-09:
This is an Electrum TREZOR plugin issue. Closing it here. I will follow the development in the linked issue.

closed 2015-07-09 by prusnak

Dehumanizer77, 2015-07-09:
Sorry I meant to open the issue in the electrum-ltc.