trezor/trezor-mcu #40: Change outputs with PAYTOSCRIPTHASH script type

Kefkius, 2015-08-21:
In firmware/signing.c - specifically, [here](https://github.com/trezor/trezor-mcu/blob/master/firmware/signing.c#L363) - there's nothing that sets `is_change` if the output's script type is `PAYTOSCRIPTHASH`.

Is it possible to use P2SH addresses as change addresses?

prusnak, 2015-08-21:
Yes. When you want to use P2SH as a change you use OutputScriptType_PAYTOMULTISIG and TxOutput with filled multisig field.. TREZOR needs to be able to construct the output by itself to verify whether an output is indeed a change (belongs to TREZOR). When you provide only P2SH it is not possible to verify this.

See https://github.com/trezor/trezor-mcu/blob/master/firmware/signing.c#L370

closed 2015-08-21 by prusnak

Kefkius, 2015-08-21:
Thanks for the very quick response @prusnak