trezor/trezor-mcu #45: allow core entropy entry through pin pad

tphyahoo, 2015-10-19:
Rather than trust host computer for source of entropy, allow user to generate entropy via alternative means (eg diceware, flipping coins, or however).

The entropy would be entered via the scrambled pin mechanism. If we just use digits 1-8, 256 bits would be 32 entries, or 128 bits would be 16. A bit of a pain, but not too bad. The entropy entry could be repeated to ensure no mistakes.

(For stylepoints, trezor could add a "hexadecimal pin pad" to slyly hint that only 3l1te experts should be using this feature.)

After the entropy entry the 3*x word passphrase would be calculated and displayed to user as usual. The user could can now either

-- throw away the entropy seed, and do future restores with current mechanism
-- or keep the entropy seed, and do future restores with extra secure mechanism of re-entering same entropy via keypad, so even if host computer is compromised the original entropy is maintained, and we don't leak down to 24 factorial, or however many if an attacker gains a scrambled word list.

As a side note on naming, I would refer to "core entropy seed" described above, versus "stretched entropy seed" which is encoded by the pass phrase that is usually referred to as the seed.

For everyone who has ever wondered if /dev/random is back doored on their system, the trezor now presents an irresistible value proposition.

originally submitted to https://github.com/trezor/python-trezor/issues/44

nullbio, 2017-06-18:
This is actually the only reason I haven't bought a Trezor yet.

saleemrashid, 2017-06-18:
@nullbio You could provide half the entropy via the computer. If you really want this feature, what entropy source are you going to use? A dice?

nullbio, 2017-06-18:
@saleemrashid Yeah, dice or something of that kind.

mokshasoft, 2017-12-13:
I don't know how the device generates entropy, but one thing that could be added is to be able to generate it by clicking the two buttons on the trezor. So the first time you set it up you would have to click the buttons for a minute or so.

prusnak, 2019-04-15:
Nobody picked that up

closed 2019-04-15 by prusnak