trezor/trezor-mcu #46: use static ram for entropy source?

tphyahoo, 2015-10-19:
Can trezor wallet copy idea from mycelium entropy and use static ram for entropy source?  This would be another way of avoiding dependence on /dev/random on naughty computer.

From https://mycelium.com/assets/entropy/me.html : 

"When Mycelium Entropy is inserted into a USB port, it takes the contents of its microcontroller’s built-in static RAM as the primary source of entropy. This method is based on the paper by D.E. Holcomb, W.P. Burleson, and K. Fu: Power-up SRAM State as an Identifying Fingerprint and Source of True Random Numbers (IEEE Transactions on Computers, vol. 58, issue 9, 2009; DOI: 10.1109/TC.2008.212). Their results were confirmed by analysing the raw data collected from the Mycelium Entropy microcontroller.

The paper gives a conservative estimate that 32 bits of SRAM contain at least one bit of entropy on power-up. (The actual ratio is closer to 22:1.) The device collects 24 kbytes of raw data. This is estimated to have more than 6144 bits of entropy, or 24 times more than required.

Multiple health checks are run on the SRAM data to confirm its randomness."

abstract at 

http://citeseerx.ist.psu.edu/viewdoc/summary?doi=10.1.1.164.6432 

and full paper at 

http://citeseerx.ist.psu.edu/viewdoc/download;jsessionid=795F927399F2A2B140FBDBCDC009BDEE?doi=10.1.1.164.6432&rep=rep1&type=pdf

prusnak, 2015-10-19:
Unlike Entropy TREZOR has a true RNG source which is based on an analogue circuit, so there is no need to perform this on our microcontroller.

Also implementing SRAM RNG and then putting the whole circuit in a Faraday cage seems like a crypto snake oil to me.

closed 2015-10-19 by prusnak

tphyahoo, 2015-10-19:
RNG source is connected computer right?

prusnak, 2015-10-19:
No.

Links:

  1. https://github.com/trezor/trezor-mcu/issues/46#issuecomment-149318863

tphyahoo, 2015-10-19:
Got it.  Sorry about the misunderstanding.

https://www.reddit.com/r/Bitcoin/comments/2d36lb/is_the_trezor_seed_entropy_truly_and_provably/

explains this well.

tphyahoo, 2015-10-19:
PS trezor is awesome :)