trezor/trezor-mcu #47: Docker Deterministic Build Fails

aussiehash, 2015-10-22:
Using OSX Kitematic, the build fails. There is no /output/ folder or binary generated

    bash-4.3$ git tag
    v1.0.0
    v1.1.0
    v1.2.0
    v1.2.1
    v1.3.0
    v1.3.1
    v1.3.2
    v1.3.3
    v1.3.4
    bash-4.3$ git checkout v1.3.4
    Note: checking out 'v1.3.4'.
    
    You are in 'detached HEAD' state. You can look around, make experimental
    changes and commit them, and you can discard any commits you make in this
    state without impacting any branches by performing another checkout.
    
    If you want to create a new branch to retain commits you create, you may
    do so (now or later) by using -b with the checkout command again. Example:
    
      git checkout -b <new-branch-name>
    
    HEAD is now at db93a50... update trezor-crypto
    bash-4.3$ ./firmware-docker-build.sh v1.3.4
    Sending build context to Docker daemon 1.333 MB
    Step 0 : FROM ubuntu:14.04
     ---> 0a17decee413
    Step 1 : ENV GCC_ARM_VERSION 4.9.3.2015q2-1trusty1
     ---> Running in c027487a9a52
     ---> 2ed75ee62628
    Removing intermediate container c027487a9a52
    Step 2 : RUN apt-key adv --keyserver keyserver.ubuntu.com --recv-keys FE324A81C208C89497EFC6246D1D8367A3421AFB
     ---> Running in 226c1723c603
    Executing: gpg --ignore-time-conflict --no-options --no-default-keyring --homedir /tmp/tmp.a19MRlw41Y --no-auto-check-trustdb --trust-model always --keyring /etc/apt/trusted.gpg --primary-keyring /etc/apt/trusted.gpg --keyserver keyserver.ubuntu.com --recv-keys FE324A81C208C89497EFC6246D1D8367A3421AFB
    gpg: requesting key A3421AFB from hkp server keyserver.ubuntu.com
    gpg: key A3421AFB: public key "Launchpad PPA for Terry Guo" imported
    gpg: Total number processed: 1
    gpg:               imported: 1  (RSA: 1)
     ---> 4db2e7600427
    Removing intermediate container 226c1723c603
    Step 3 : RUN gpg --keyring /etc/apt/trusted.gpg --primary-keyring /etc/apt/trusted.gpg --fingerprint FE324A81C208C89497EFC6246D1D8367A3421AFB
     ---> Running in 1742e18f0173
    gpg: directory `/root/.gnupg' created
    gpg: new configuration file `/root/.gnupg/gpg.conf' created
    gpg: WARNING: options in `/root/.gnupg/gpg.conf' are not yet active during this run
    gpg: keyring `/root/.gnupg/pubring.gpg' created
    gpg: /root/.gnupg/trustdb.gpg: trustdb created
    pub   1024R/A3421AFB 2012-09-03
          Key fingerprint = FE32 4A81 C208 C894 97EF  C624 6D1D 8367 A342 1AFB
    uid                  Launchpad PPA for Terry Guo

     ---> b7bb499ad881
    Removing intermediate container 1742e18f0173
    Step 4 : RUN echo "deb http://ppa.launchpad.net/terry.guo/gcc-arm-embedded/ubuntu trusty main" >> /etc/apt/sources.list && apt-get update
     ---> Running in 663a81663d04
    Ign http://ppa.launchpad.net trusty InRelease
    Ign http://archive.ubuntu.com trusty InRelease
    Get:1 http://ppa.launchpad.net trusty Release.gpg [316 B]
    Get:2 http://archive.ubuntu.com trusty-updates InRelease [64.4 kB]
    Get:3 http://ppa.launchpad.net trusty Release [15.1 kB]
    Get:4 http://archive.ubuntu.com trusty-security InRelease [64.4 kB]
    Get:5 http://ppa.launchpad.net trusty/main amd64 Packages [508 B]
    Hit http://archive.ubuntu.com trusty Release.gpg
    Get:6 http://archive.ubuntu.com trusty-updates/main Sources [301 kB]
    Get:7 http://archive.ubuntu.com trusty-updates/restricted Sources [4513 B]
    Get:8 http://archive.ubuntu.com trusty-updates/universe Sources [176 kB]
    Get:9 http://archive.ubuntu.com trusty-updates/main amd64 Packages [799 kB]
    Get:10 http://archive.ubuntu.com trusty-updates/restricted amd64 Packages [22.7 kB]
[... 549 lines cut ...]
    Step 2 : RUN apt-key adv --keyserver keyserver.ubuntu.com --recv-keys FE324A81C208C89497EFC6246D1D8367A3421AFB
     ---> Using cache
     ---> 4db2e7600427
    Step 3 : RUN gpg --keyring /etc/apt/trusted.gpg --primary-keyring /etc/apt/trusted.gpg --fingerprint FE324A81C208C89497EFC6246D1D8367A3421AFB
     ---> Using cache
     ---> b7bb499ad881
    Step 4 : RUN echo "deb http://ppa.launchpad.net/terry.guo/gcc-arm-embedded/ubuntu trusty main" >> /etc/apt/sources.list && apt-get update
     ---> Using cache
     ---> 06c0669f6097
    Step 5 : RUN apt-get install -y build-essential git python
     ---> Using cache
     ---> f34f5f36d268
    Step 6 : RUN apt-get install -y gcc-arm-none-eabi=$GCC_ARM_VERSION
     ---> Running in 73be42ecc791
    Reading package lists...
    Building dependency tree...
    Reading state information...
    E: Version '4.9.3.2015q2-1trusty1' for 'gcc-arm-none-eabi' was not found
    The command '/bin/sh -c apt-get install -y gcc-arm-none-eabi=$GCC_ARM_VERSION' returned a non-zero code: 100
    Unable to find image 'trezor-mcu-build:latest' locally
    Pulling repository docker.io/library/trezor-mcu-build
    Error: image library/trezor-mcu-build:latest not found
    ---------------------
    Firmware fingerprint:
    ./firmware-docker-build.sh: line 19: sha256sum: command not found

jhoenicke, 2015-10-22:
It looks like you need the program `sha256sum` on your local system to check the firmware (the firmware has been built successfully, but the command that should compute the checksum failed).  Try replacing `sha256sum` in firmware-docker-build.sh with `shasum -a 256`

aussiehash, 2015-10-22:
I've created another Docker VM, tried it again, and this time a binary was produced which does not match the firmware fingerprint.

No tag produces /output/trezor-master.bin
`v1.3.4` produces /output/trezor-v1.3.4.bin
`v1.3.3` fails to build

Unfortunately the sha256 hashes do not match the official firmware
https://mytrezor.com/data/firmware/releases.json

No idea why `sha256sum`  fails

    Firmware fingerprint:
    ./firmware-docker-build.sh: line 19: sha256sum: command not found
    bash-4.3$ cd output
    bash-4.3$ ls
    trezor-master.bin	trezor-v1.3.4.bin
    bash-4.3$ cd ..

    bash-4.3$ cat output/trezor-v1.3.4.bin | shasum -a 256
    8b6bb084410e857bcefd83ec565fb89dee93dccd515ef04e0be4a342b8b7ba89  -
    bash-4.3$ echo -n output/trezor-v1.3.4.bin | shasum -a 256
    2d545710edb43c6826f97df6ffff212551fc091f0e37f357ddb0fa1f26e920e2  -
    bash-4.3$ echo output/trezor-v1.3.4.bin | shasum -a 256
    581bc9243e804284e40e8befa8581d3ec4ed24d0f82ef081bf218cc3f6643c52  -
    bash-4.3$ echo -n output/trezor-v1.3.4.bin | openssl dgst -sha256
    2d545710edb43c6826f97df6ffff212551fc091f0e37f357ddb0fa1f26e920e2
    bash-4.3$ cat output/trezor-v1.3.4.bin | openssl dgst -sha256
    8b6bb084410e857bcefd83ec565fb89dee93dccd515ef04e0be4a342b8b7ba89
    bash-4.3$ cat -n output/trezor-v1.3.4.bin | openssl dgst -sha256
    64a15b3e8f876af8d715afb7fe72a159d129aafd3a06d96cf1310b4861f595b9

    bash-4.3$ cat output/trezor-master.bin | shasum -a 256
    b69d9067edfc685f3a28627c5ced98458ebc50b40bd9879ceba07d223caeba11  -
    bash-4.3$ echo -n output/trezor-master.bin | shasum -a 256
    336cf1e914276c0bf031ad55537ee742555a137a12adc8ec569197331f956a1d  -
    bash-4.3$ echo output/trezor-master.bin | shasum -a 256
    91e3dffc42ed1686f623f4f2ac545eea9e6673c49f35e3d450045d3c5e37731a  -
    bash-4.3$ echo -n output/trezor-master.bin | openssl dgst -sha256
    336cf1e914276c0bf031ad55537ee742555a137a12adc8ec569197331f956a1d
    bash-4.3$ cat output/trezor-master.bin | openssl dgst -sha256
    b69d9067edfc685f3a28627c5ced98458ebc50b40bd9879ceba07d223caeba11
    bash-4.3$ cat -n output/trezor-master.bin | openssl dgst -sha256
    12ad2c231130e1bec6846f094cfaf47696f9c23504863b3de59ac8b38d21e73a

jhoenicke, 2015-10-22:
What does `arm-none-eabi-gcc --version` print with your docker image?

I get the correct checksum with both 4_9-2015q2 and 4_9-2015q3:

```
./firmware-fingerprint.sh  ~/Downloads/trezor-1.3.4.bin 
---------------------
Firmware fingerprint:
49e044eec84a9c210a09319d27a3ab8ba889ddeaa4d68f99d163f65267fce134  -
>sha256sum firmware/trezor.bin 
49e044eec84a9c210a09319d27a3ab8ba889ddeaa4d68f99d163f65267fce134  firmware/trezor.bin
```

aussiehash, 2015-10-22:
I can't figure out where Docker has installed `arm-none-eabi-gcc`

It's not in `/usr/local/`

no idea where docker put `/libopencm3` either

jhoenicke, 2015-10-22:
Try replacing 2015q2 by 2015q3 in the Dockerfile.  I think the problem is that 2015q2 is no longer available through apt-get install, after 2015q3 was  released last September.  Fortunately, the changes are so small that they will both produce the same binary.

Unfortunately, this is not a general solution to the problem. When Terry Guo's releases a new arm compiler the previous one is removed from the ubuntu servers and thus the deterministic build process will fail.

fikevihu, 2015-10-22:
@jhoenicke, thank you!  Your suggestion resulted in the expected signature being built for v1.3.4.

I simply changed the following in Dockerfile:
```
RUN apt-get install -y build-essential git python gcc-arm-none-eabi
```

To:
```
RUN apt-get install -y build-essential git python
RUN apt-get install -y curl lib32z1
RUN curl -SL https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q2-update/+download/gcc-arm-none-eabi-4_9-2015q2-20150609-linux.tar.bz2 | tar -xjf -
ENV PATH /gcc-arm-none-eabi-4_9-2015q2/bin:$PATH
```

aussiehash, 2015-10-23:
>Try replacing 2015q2 by 2015q3 in the Dockerfile. 

Thanks @jhoenicke, success with building v1.3.4 ! My steps were the following :

* Launch Kitematic
* File > Open Docker Command Line Terminal
* `mkdir docker`
* `cd docker`
* `git clone https://github.com/trezor/trezor-mcu.git`
* `cd trezor-mcu`
* `git checkout v1.3.4`
* `vi Dockerfile` (and replace 2015q2 with 2015q3)
* `docker build -t trezor-mcu-build .`
* `./firmware-docker-build.sh v1.3.4`

    Firmware fingerprint:
    ./firmware-docker-build.sh: line 19: sha256sum: command not found
    bash-4.3$ cat output/trezor-v1.3.4.bin | shasum -a 256
    49e044eec84a9c210a09319d27a3ab8ba889ddeaa4d68f99d163f65267fce134  -

prusnak, 2015-12-06:
I just opened an issue to replace current Docker build with something less exotic, e.g. gitian. Any help is welcome! https://github.com/trezor/trezor-mcu/issues/66

closed 2015-12-06 by prusnak