trezor/trezor-mcu #54: Display mBTC/bits

prusnak, 2015-11-13:


bitsol, 2018-01-20:
Display before signing?
Should this be configurable?

bitsol, 2018-01-20:
https://github.com/trezor/trezor-mcu/blob/cd763b979b0001d4e23fffaf68fa59cfe483a6ba/firmware/layout2.c#L190
Should the change be implemented in layoutConfirmTx(), or a "lower-level" function?
In other words, count me in.

prusnak, 2018-01-20:
Should be configurable, the preferred format should be stored in Storage.

Yes, `layoutConfirmTx` should be a good place to perform the change.

bitsol, 2018-01-20:
Sorry, I'm not really familiar with the organization of the source tree. Where should I define the enum, or for that matter, any type of configuration variable?

bitsol, 2018-01-20:
I just realized how easy it is to get around the `storage.c` file.
https://github.com/trezor/trezor-mcu/blob/bd660655ee26711bcba8d25d6555cc8246e7e2df/firmware/storage.c#L490
This is the kind of function I would write, and then read that value in `layoutConfirmTx`. Is that good?

saleemrashid, 2018-01-20:
I actually have a branch offline that does this (except for the `Storage` changes). Shall I push that up?

bitsol, 2018-01-21:
@prusnak thoughts?
I didn't get here fast enough I guess...

prusnak, 2018-01-21:
@bitsol please check pull request by Saleem here: https://github.com/trezor/trezor-mcu/pull/292

Closing this issue in favor of that PR. Let's continue discussion there.

closed 2018-01-21 by prusnak