trezor/trezor-mcu #59: Test recovery with initialized device

prusnak, 2015-11-13:
* same as standard recovery from user's point of view
* device will not store the seed, but shows "Recovery seed checked." // "Recovery seed mismatch."

Feature to make sure, that the seed works. Will be used before wiping the device before sending it for RMA etc.

ondra-novak, 2016-10-23:
BUMP

Please add as "verify recovery" Currently I have device at risk  because I am not sure, that I have correct sheet with the words (and which one is correct, because I have more such sheets). This can be solved only by moving all funds to other wallet, wipe device, recreate the seed and move funds back to the Trezor.. With ability to verify recovery I can skip this annoying procedure.

prusnak, 2016-11-09:
Other option perform the check on another TREZOR device.

romanz, 2016-12-31:
What do you think about the following solution?

https://github.com/trezor/trezor-common/compare/master...romanz:dry-run
https://github.com/trezor/trezor-mcu/compare/master...romanz:dry-run
https://github.com/trezor/python-trezor/compare/master...romanz:dry-run

The changes above add a new command for running the new Matrix-based recovery process in "dry run" mode, only to see if the mnemonic is correct.

romanz, 2016-12-31:
P.S.
Feel free to ignore and re-generate the protobuf-related commits at trezor-mcu and python-trezor.

jnicolau, 2017-05-01:
Is this feature on the make. Currently there is no way to test the seed? Any workaround?

romanz, 2017-05-26:
@prusnak Ping :)
(let me know if it's not the intended way for implementing this feature...)

prusnak, 2017-05-29:
Will review it soon.

jhoenicke, 2017-05-29:
I thought about a different feature that would also allow to check the recovery seed.  Have only one new message that exchanges the current configuration with another one stored at an offset in the configuration sector.  It is like switching between two different Trezors.  This also allows for securely updating to a new seed without the need to recover twice.  And it has other use cases such as having one seed for private use and one seed for business use.

Your solution is in some sense simpler since it only does one thing.  On the other hand the solution with multiple configurations, despite being more versatile, doesn't require any changes to the recovery function. Only one additional call to switch between configurations (and the code that writes to the storage sector needs to be changed to preserve the second configuration).  So it would be less or about equal amount of code in the firmware.

The disadvantage of multiple configurations would be that it requires more calls to verify the seed: get master public key, switch configuration, recover device (without PIN), get master public key, check it, wipe device (and explain to the user why he has to confirm the scary message saying that all his data will be erased), switch configuration.  This complexity can be hidden by the wallet, though, except for the wipe confirmation.   But more importantly, this only works if the second configuration is not currently used for something else.  The last point can be avoided if we hide the full power of the feature to the end user.

prusnak, 2017-05-29:
I don't like several configurations in one TREZOR. We are having really hard time to explain the passphrase concept. Having more configurations would be even more hard to understand.

saleemrashid, 2017-05-29:
@jhoenicke I implemented that before but the consensus from @prusnak and others was that it was a bad idea

prusnak, 2017-06-20:
Sorry for not reviewing this earlier, had so many other things on my plate.

I think this is way to go, but I would change some things:

a) Let's not introduce new message, just add new field to RecoveryMessage (dry_run sounds OK)
b) I would change the check at the end, so it has 3 possible outcomes:
* the mnemonic is invalid
* the mnemonic is valid, but not the same as the one in device
* the mnemonic is valid and is the same as the one in device

romanz, 2017-06-20:
Both suggestions sound good :)
Will update the PR soon...

prusnak, 2017-07-12:
Implemented in https://github.com/trezor/trezor-mcu/pull/176

Great work Roman!

closed 2017-07-12 by prusnak