trezor/trezor-mcu #62: show passphrase on Trezor

prusnak, 2015-11-13:
Instead of typing the passphrase twice each time, user should enter the passphrase only once.

User should be able to show the passphrase on TREZOR instead on PC

mariodian, 2018-01-20:
Confirming that the passphrase is valid just by looking at Trezor's screens is not as safe as the necessity to enter it twice. Since people don't read strings "character by character" but instead look at chunks of strings, it may be easy to overlook some missing chars that should have been there. That's a problem especially if you're creating a new wallet with a new passphrase that you may not yet be familiar with.

prusnak, 2018-01-20:
I agree with the above comment

karel-3d, 2018-01-25:
Confirming on screen is protecting against an attack, writing twice is protecting against the user. It's different.

prusnak, 2018-02-04:
Implemented in https://github.com/trezor/trezor-mcu/pull/246

closed 2018-02-04 by prusnak