trezor/trezor-mcu #63: Trezor repeatedly asks for the same transactions on sending

runn1ng, 2015-11-23:
If I am sending a transaction that spends more outputs of the same transaction, the previous transaction gets streamed to TREZOR repeatedly for every output, even when it's not necessary, which slows the signing

Extreme (but real) example:

I have received a transaction with 200 outputs, each to the same receiving address. When I want to spend the transaction now, for all of the 200 outputs, the previous transaction (which is the same) is streamed to TREZOR, with all the 200 outputs! So instead of necessary 200 outputs streamed to TREZOR, I am now streaming 200*200=40.000 outputs, over and over again.

TREZOR should remember, which previous transactions were "checked" already.

runn1ng, 2015-11-23:
This can also be used as an "attack" against TREZOR users (wallets will use the freshest transactions, which will be "unspendable" during the high time neccessary waiting time)

jhoenicke, 2015-11-23:
It's not easy to fix. It complicates the signing protocol, which is critical for the security of the Trezor.  For the normal case, it is quite unlikely that one receives several coins in the same transaction. And even if this is fixed, an "attack" can still be achieved by sending 200 transactions with 200 outputs to 200 different Trezor owners, where every transaction contains one output for each receiver.

I think an easier to implement optimization is to compute several hashes for different input at the same time, to reduce the quadractic part of the signing algorithm.

runn1ng, 2015-11-23:
I agree that complicating the signing function is not a good idea.

The multiple owners attack is slighly harder to do and orchestrate than just sending one transaction with 200 outputs. But yeah it is essentially the same thing.

Your optimisation sounds better, yeah.

prusnak, 2015-11-24:
Unfortunately, the real fix is to implement SIGHASH_WITHINPUTVALUE within the Bitcoin protocol, so we can get rid of streaming prevtxs completely. Everything else is just a workaround of one very importang thing Satoshi missed. :-/

Filed here and here:
https://github.com/bitcoin/bitcoin/issues/7089
https://github.com/bitcoinxt/bitcoinxt/issues/13

jonathancross, 2016-02-23:
Unfortunately both of those issues are now closed.
On the bright side, it seems SegWit may allow easy upgrades such as `SIGHASH_WITHINPUTVALUE` from what I have read here: http://lists.linuxfoundation.org/pipermail/bitcoin-dev/2015-December/011938.html

prusnak, 2016-02-23:
@jonathancross I've understood that SegWit already includes value (although hashed) in the input script, so the issue is solved even without any further updates.

Closing this as this will be no longer an issue once TREZOR can speak Segwit (and it will no longer need to stream previous transactions).

closed 2016-02-23 by prusnak