trezor/trezor-mcu #64: Put STAGE_2 after confirmations

runn1ng, 2015-11-24:
The "STAGE_2" streaming can take a long time, especially with really big previous transactions. Would it be possible to put the STAGE_2 after the confirmations? So rather than "wait a long time - confirm - wait a long time", the scenario would be "confirm - wait a long time".

This might probably not be possible, I am just interested why :)

prusnak, 2015-11-24:
There are two types of confirmations. Confirm Output and Confirm Tx (with fee). The latter could not be done before all prevtxs are streamed in (that's the whole purpose of stage 2 - to calculate the fee which is then shown in Confirm Tx dialog). Or is there something I am missing?

runn1ng, 2015-11-24:
ok

closed 2015-11-24 by runn1ng