trezor/trezor-mcu #67: Initialize while waiting after wrong PIN

runn1ng, 2015-12-22:
When TREZOR is waiting for PIN matrix after repeated wrong PIN attempts, and computer stops communication, starts another communication and sends Initialize, TREZOR replies with PIN request and starts counting down again, even when Initialize shouldn't get PIN request.

Also, maybe more importantly, even when the computer correctly sends the PIN, the countdown starts *again*, as if the previous request didn't run.

Examples, using python-trezor and forking

* https://gist.github.com/runn1ng/c29dfc1e3c778c7a94bb
* https://gist.github.com/runn1ng/172ad85a5b6613f2b435

jhoenicke, 2016-05-28:
I analyzed it a bit.  One problem is that usbTiny is not set while waiting.  The init request is processed as usual but recursively inside the pin function, which is bad.  

I think the cleanest fix is to set usbTiny in the wait loop and when init is received abort the loop and return false (also call fsm_sendFailure which will answer the init request).

prusnak, 2016-05-28:
Fixed in #88

closed 2016-05-28 by prusnak