trezor/trezor-mcu #68: Initialize while listing seeds for device setup doesn't return features

runn1ng, 2015-12-23:
(similar to issue #67)

When the device is listing seed words in device setup, computer should be able to reset the device with Initialize message.

It doesn't work and the Initialize never returns any features. The computer is expecting them and hangs.

Example in python, similar to examples in #67 
* https://gist.github.com/runn1ng/d4cbd29452ed81501c39

(run with *un*initialized device)

runn1ng, 2015-12-23:
The same bug is *not* present in device recovery

jhoenicke, 2016-07-04:
Isn't a `fsm_sendFailure` in `reset.c` line 146 (between layoutHome() and return) enough to solve this issue?

prusnak, 2016-07-04:
Good find, Jochen! Thanks! Fixed in https://github.com/trezor/trezor-mcu/commit/633024a9933c2f6b672ac6ec3e7e69c384ec90be

closed 2016-07-04 by prusnak