trezor/trezor-mcu #69: Protocol failure message - how to recover?

kyuupichan, 2016-01-17:
I'm trying to implement wallet recovery in electrum.

Electrum currently bombs out if it gets a Failure message when recovering a wallet when going through the trezor recovery steps of entering 24 words.  If I enter a misspelt word or the wrong word, I get a  "Wrong word retyped" or "Word not found in wordlist" error.  Protcol-wise, what should I do to say "OK" and be able to continue?  The current loop in client.py, call() around line 164 is raising an exception via callback_Failure.  I would like to be able to re-attempt that word.  Is this possible?  I really hope it is - simply abandoning the user after they've entered 20 words and mistyped the 21st is not very friendly.

In a similar vein,  if the user cancels their pin entry, I get an exception.  I'd like to be able to re-implement callback_Failure and return an "OK; ignore it" message.  At present, in order to not bother the Electrum user with unwanted exceptions, I'm overriding callback_Failure and throwing a SilentException (defined in Electrum) in order to unwind the stack.  The caller in Electrum catches the exception and displays it to the user, unless it's a SilentException, in which case it's just swallowed.  This works but is a tedious workaround to the fact that the call() function / the protocol seems to require an exception to be thrown in case of failure because there is no other way of unwinding the stack because there is no "OK; failure acknowleged" protocol message that the device will accept after it sends a failure message.  Am I missing something?  I hope my question is clear.

prusnak, 2016-01-17:
Currently this is not possible, because TREZOR - hardware - aborts the whole sequence and you need to start from the beginning.

closed 2016-01-17 by prusnak

kyuupichan, 2016-01-17:
Thanks Pavel.  Don't you think that would be a reasonable firmware enhancement?

prusnak, 2016-01-18:
I do. But currently we don't have capacity to implement this. It adds
too much complexity and the gain is relatively small (you don't do
recovery every day as opposed to other operations). The firmware is open
source and anyone can add this feature.

 

Links:

  1. https://github.com/trezor/trezor-mcu/issues/69#issuecomment-172393597

jhoenicke, 2016-01-18:
BTW, you can also check if the word is in the word list on the client site.  This is how the web interface avoids this problem.

kyuupichan, 2016-01-18:
@jhoenicke Fair enough, but it's some effort for a 30% solution; so not very interesting to me.  My only complaint is the user-unfriendliness of forcing them to start over from the beginning.

kyuupichan, 2016-01-18:
@jhoenicke and even then - what are the languages available?  How do I  know I'm using the same language as the Trezor?  The undocumented 'language' feature on trezors I've seen so far has been either an empty string or lower-case English.  I cannot reasonably validate their input unless I know what language they are using and what languages and dictionaries the trezor accepts.  To the best of my knowledge the latter is undocumented and currently just "english" for which blank is a synonym.  But if Trezor accepted Spanish or Japanese, or *anything else* I haven't implemented 100% perfectly, I'm essentially denying the user the ability to recover their wallet.  So no, this is not a way forward, unless I'm missing something very basic.

prusnak, 2016-01-18:
I was never a fan of wordlist that was not English and TREZOR does not support recovery of non-English wordlists.