trezor/trezor-mcu #70: Decryption doesn't work for non-trezor encrypted messages

kyuupichan, 2016-01-17:
I get a bad nonce error about decrypting a message encrypted with Electrum in its encrypt message screen.  

I believe Electrum does the same as bitcoind so I'd like to understand what's wrong here.

Thanks.

prusnak, 2016-01-17:
A particular example of such message would help ...

Also I don't believe that bitcoind does support message encryption.

Aren't you mixing message signing with encrypting of the message?

kyuupichan, 2016-01-18:
Hi Pavel,

In terms of bitcoind support I think I was mistaken.

Just fire up electrum (HEAD - older versions raise an unimplemented exception for decryption only) with a Trezor wallet.  Right-click and address and encrypt anything, and then try to decrypt it from the same dialog box.  Note the encryption is implemented by Electrum (and has been for a long time with no user complaints) and only uses public keys, so because Electrum has those for the wallet it does it itself.  The decryption uses the hardware device with the private keys and fails.

I have no interest in who is wrong or right here; I just want it to work.  Unfortunately it seems there are few independent implementations of this, as you essentially pointed out, to verify against.

Concretely, encrypting the unquoted word "Test" with public key 03fbbc567505a66b96f6b64b453f2c9105732e2b9daf1e7d2ba0bddd8a20edd2ca
on Electrum gave me (of course it changes each time):

QklFMQJw1i2s1Pw9BqFxNGUFSsMPimtrTf80xkvqmGjTt8j3UcbjPYKNX8xK/6rYrOFvBD0bps0eSDcEeGFe7kua5jKGGgOxMJ2RvfjjScGl7rjCVw==

Pressing decrypt and the trezor client software throws an exception:

  File "/Users/neil/src/electrum/gui/qt/util.py", line 545, in run
    result = task.task()
  File "/Users/neil/src/electrum/plugins/trezor/plugin.py", line 127, in decrypt_message
    result = client.decrypt_message(address_n, nonce, message, msg_hmac)
  File "/Users/neil/src/electrum/plugins/trezor/clientbase.py", line 185, in wrapped
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/trezorlib/client.py", line 84, in wrapped_f
    ret = f(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/trezorlib/client.py", line 502, in decrypt_message
    return self.call(proto.DecryptMessage(address_n=n, nonce=nonce, message=message, hmac=msg_hmac))
  File "/usr/local/lib/python2.7/site-packages/trezorlib/client.py", line 127, in call
    msg = handler(resp)
  File "/Users/neil/src/electrum/plugins/trezor/clientbase.py", line 31, in callback_Failure
    raise RuntimeError(msg.message)
RuntimeError: Invalid nonce provided

Neil

prusnak, 2016-02-23:
Can you tell me what exactly are the parameters you are feeding to `client.decrypt_message` function? My guess is there is a problem with encoding of `nonce` (using hex string instead of binary string or something similar).

kyuupichan, 2016-02-24:
Decryption isn't the problem - I encrypted a message from the command line trezorctl and put it in Electrum, which decrypted it just fine using the trezor.

However, the encrypted message produced by Electrum cannot be decrypted by the trezor.  So for some reason the encrypted message format is different; I have no idea what Trezor is expecting but we've never had a complaint about Electrum's encrypted messages.

kyuupichan, 2016-02-24:
Electrum doesn't use the device to encrypt; it has its own code.  I could bypass that but I want to understand before doing something like that.

kyuupichan, 2016-02-24:
Here's the trezor encrypted payload:

'payload': 'A5Lh6HY3491cF3pKpG8ZT80q7cdK2X+puxPVb3gYDoGvEI0jEi2izdC6xmEKRE4='

Here's Electrum's:

QklFMQIQRYQJ56bY6Jg/KjK1b2nbctnxwoF9yA6l7/K7IG2Bn/2Eg5k3A3QfwqLPKFWxg0pdKq5hDjzFewx9FAR7vVW5rpp03RAaLW4havHZ0d6UEw==

Electrum's is longer, doesn't use + signs, and always ends in ==

kyuupichan, 2016-02-24:
Please try it for yourself; it only takes a couple of minutes

kyuupichan, 2016-02-24:
I think I know what the issue is - the Electrum ciphertext includes magic bytes and the hmac.  Let me see if I can figure this out...

kyuupichan, 2016-02-24:
Pavol - it seems the HMAC generation is fundamentally incompatible between Electrum and the h/w.  This means that the h/w can't decrypt an electrum-encrypted message.  It also means a restored electrum h/w wallet (i.e. Electrum software wallet with no hardware)  is encrypt / decrypt incompatible with the Electrum h/w wallet, even if we switch to Trezor's method.

Is there a BIP for encryption, or did you make it up like we did for Electrum?

prusnak, 2016-02-24:
I wanted to create [SLIP-0012](https://github.com/satoshilabs/slips/blob/master/slip-0012.md) which will document this, but never got to writing this down. 

Now, I remember why I did the change. I wanted to introduce a display_only flag, specified by the one who encrypts the message. And when this is set to True, TREZOR shows message just on the display, but does not send it to a computer (=untrusted environment). I think it really makes sense.

kyuupichan, 2016-02-24:
Did you get peer review of your method?  We would consider switching Electrum's method to be identical, but we are sceptical of unreviewed crypto methods.

prusnak, 2016-02-24:
Was Electrum method peer reviewed? I think the change is minimal, but I don't remember the details about the differences now.

kyuupichan, 2016-02-24:
Thomas had Greg M look at it; he said it was initially insecure.

kyuupichan, 2016-02-24:
But I'll take your response as a "No" :)

prusnak, 2016-05-17:
I decided to remove ECIES from the firmware for the time being. There is no common standard, maybe it will change in the future.

closed 2016-05-17 by prusnak

kyuupichan, 2016-05-17:
I see Pavel.  Does that mean you've disabled encryption?  Won't that be an issue for those who might have encrypted something with it?

prusnak, 2016-05-17:
I don't know of anyone using this feature. If there is any, she can use the older firmware (1.3.5) to decrypt the stuff.

jhoenicke, 2016-05-18:
I looked over the differences.  The Trezor implementation is based on this
https://web.archive.org/web/20160210174934/http://memwallet.info/btcmssgs.html
I think the only difference between this and Trezor's implementation is that Trezor uses the high bit of the first byte of the message as "only show on Trezor" flag (the first byte indicates whether message is signed or not).

The electrum implementation is quite different and probably developed independently. Both implement basically ECIES, but

- Trezor uses PBKDF with salt "Bitcoin Secure Message", iter=2048; electrum uses a simple SHA512 as KDF.
- Trezor uses aes256-cfb, Electrum uses aes128-cbc
- The deserialization of the KDF result into enckey, iv, hmackey is slightly different: Trezor: enckey||hmackey||iv; electrum: enckey||iv||hmackey   (and enckey is shorter for aes128).
- The encrypted message in trezor is structured: first byte sign_flag, then varint message len, followed by message, optionally followed by signing address and signature.  In Electrum the message is just the encrypted message and signing is not supported.
- For electrum the base64 encoded message starts with a 4-byte version string.  

I don't see a security problem with any of the two algorithms.  PBKDF is a bit overkill, KDF1 or just using one iteration would be sufficient.  SHA512 as KDF has the disadvantage that it is not extendable (e.g. for aes256).  Structuring the message has the advantage of being able to include the "only show on trezor" flag and the optional signature.  Having the 4 byte version at the beginning like electrum is, of course,  also advantageous.

jhoenicke, 2016-05-18:
So what is the long-term solution? I think some kind of encryption/decryption) is useful, e.g., for password managers.  Since this is asymmetric encryption, it makes sense if it works for more than one wallet.

Possibilities:
- implement Electrum's method in Trezor, optionally supporting decryption for both old and new format.
- implement Trezor's method in Electrum.
- implement a new method, e.g. similar to the old Electrum's method but with a new version number and a structured plaintext message that allows signatures or the 'never-leave-hardware-device' flag.  It can be signalled by a new version string.

I think it is enough to support only the new method for encryption.  For backward compatibility, decryption can work also for the old method.

prusnak, 2016-05-18:
I backpedalled on the current implementation, because it did not receive enough peer-review attention (same applies for Electrum method).

Now that we support ed25519 I think we should try to implement crypto_box (Curve25519-XSalsa20-Poly1305): https://nacl.cr.yp.to/box.html or similar 25519-based public-key scheme. Although that is much more work (because currently we don't support XSalsa20 and Poly1305).

kyuupichan, 2016-05-18:
I think some kind of documented standard that isn't broken, even if imperfect, is good.  Otherwise we see people do their own thing, back them out, and some like me worry about users that might have committed something, so will end up having to support many methods.  It'd be good to have at least one we can converge on that has been vetted by at least one or two experts.  I don't really care what the method is, other than avoiding unnecessarily importing new libraries and/or adding lots of new code.  Unfortunately it sounds like 25519 is going that way.

ecdsa, 2017-03-02:
Electrum 2.8 will encrypt wallet files using ECIES, with a keypair derived from the password. 
it would be nice to have this feature with hardware wallets as well.

prusnak, 2017-03-02:
I suppose you have not changed your implementation to a peer reviewed one, have you?

ecdsa, 2017-03-02:
I have not changed Electrum's implementation recently. I did change it some time ago, after @gmaxwell reviewed it and found several issues. I am willing to change it again if you have constructive comments about it.

However, I am not willing to delay this feature indefinitely without proper criticism, because:
 - Users receiving bitcoins on a wallet they do not own is currently the #1 cause of Bitcoin losses with Electrum. Having to type their password to open the wallet will protect them against that.
 - We are using ECIES to encrypt public information (addresses, etc), not private keys. If it turns out the crypto is broken, we will be in the same situation as today. 
 - Encryption will protect users against someone replacing their wallet file, or replacing the xpub/addreses in their wallet file. It will not protect the wallet when it is loaded in RAM.