trezor/trezor-mcu #85: Add ECDH support for GPG public-key decryption

romanz, 2016-05-16:
As an experimental patch, https://github.com/trezor/trezor-mcu/commit/ecfdc0dd29f020d9fcdfe95a70483b4f5961efcf seems to be working with NIST256P curve. 
In order to test it, I have added the following (very hackish - and should be rewritten) Python code to the [ECDH branch](https://github.com/romanz/trezor-agent/compare/ecdh):

 - export GPG subkey supporting encryption: https://github.com/romanz/trezor-agent/commit/968c8bafccd5072e9bd3d2be0b9ea3e954776452
 - use "gpg-agent" emulation for public-key decryption: https://github.com/romanz/trezor-agent/commit/11ccc86618572548dc34267ad24bdac915abfa42

Also, I guess it would be nicer to add a separate API for this operation (since it's not really related to the "SignIdentity" API... I have added it there only to make the prototyping easier).

romanz, 2016-05-16:
Demo usage:

```

$ export TREZOR_GPG_USER_ID="testing"

$ trezor-gpg create -s > /tmp/key 
2016-05-16 21:22:16,897 INFO       nist256p1 GPG public key <8D17941D1414BD9C> created at 2016-05-16 21:22:13 for "testing"
2016-05-16 21:22:17,030 INFO       adding subkey to primary GPG key "testing" (A49CBFEE317EDBEF)
2016-05-16 21:22:17,030 INFO       confirm signing subkey with gpg-agent
2016-05-16 21:22:17,178 INFO       signing digest: D7459D69D2B6DD47BABA8AA4B2A6AB955094E3A0BBC00E3F1F790F65B57B3277

$ gpg2 --import /tmp/key 
gpg: key 317EDBEF: "testing" 1 new signature
gpg: key 317EDBEF: "testing" 1 new subkey
gpg: Total number processed: 1
gpg:            new subkeys: 1
gpg:         new signatures: 1

$ gpg2 -k
/home/roman/.gnupg/pubring.kbx
------------------------------
pub   nistp256/317EDBEF 2016-05-14 [SC]
uid         [ultimate] testing
sub   nistp256/3ECC4691 2016-05-16 [E]

$ python -m trezor_agent.gpg.agent &
[1] 6788

$ 2016-05-16 21:22:51,916 DEBUG      connected to Trezor F83618F42686D73F8D964F8F                                                         factory.py:27
2016-05-16 21:22:51,916 DEBUG      label    :                                                                                           factory.py:28
2016-05-16 21:22:51,917 DEBUG      vendor   : bitcointrezor.com                                                                         factory.py:29
2016-05-16 21:22:51,917 DEBUG      version  : 1.3.5                                                                                     factory.py:33
2016-05-16 21:22:51,917 DEBUG      revision : e1fa896d6c9e9027e95d0b8b31032fcd943dd5ba                                                  factory.py:34
2016-05-16 21:22:51,945 INFO       waiting on /home/roman/.gnupg/S.gpg-agent                                                            agent.py:167

$ date | gpg2 --encrypt -r "testing" | gpg2 --decrypt
2016-05-16 21:23:02,600 INFO       got: 'RESET'                                                                                         agent.py:172
2016-05-16 21:23:02,600 INFO       got: 'OPTION ttytype=xterm'                                                                          agent.py:172
2016-05-16 21:23:02,600 INFO       got: 'OPTION display=:0'                                                                             agent.py:172
2016-05-16 21:23:02,601 INFO       got: 'OPTION xauthority=/home/roman/.Xauthority'                                                     agent.py:172
2016-05-16 21:23:02,601 INFO       got: 'OPTION putenv=DBUS_SESSION_BUS_ADDRESS=unix:abstract=/tmp/dbus-PVV9Y73qx2,guid=41ebd5d35db6f79a8d4cb046573970eb' agent.py:172
2016-05-16 21:23:02,601 INFO       got: 'GETINFO version'                                                                               agent.py:172
2016-05-16 21:23:02,601 INFO       got: 'OPTION allow-pinentry-notify'                                                                  agent.py:172
2016-05-16 21:23:02,601 INFO       got: 'OPTION agent-awareness=2.1.0'                                                                  agent.py:172
2016-05-16 21:23:02,601 INFO       got: 'AGENT_ID'                                                                                      agent.py:172
2016-05-16 21:23:02,602 INFO       got: 'HAVEKEY 432F841C4901706947C5697C9B775C00BD17DAE4'                                              agent.py:172
2016-05-16 21:23:02,602 INFO       got: 'HAVEKEY 37E240CCD4F9BF83A7EE343798F3CB638156EA68 432F841C4901706947C5697C9B775C00BD17DAE4'     agent.py:172
2016-05-16 21:23:02,609 INFO       got: 'HAVEKEY 432F841C4901706947C5697C9B775C00BD17DAE4'                                              agent.py:172
2016-05-16 21:23:02,610 INFO       got: 'RESET'                                                                                         agent.py:172
2016-05-16 21:23:02,610 INFO       got: 'SETKEY 432F841C4901706947C5697C9B775C00BD17DAE4'                                               agent.py:172
2016-05-16 21:23:02,610 INFO       got: 'SETKEYDESC Please+enter+the+passphrase+to+unlock+the+OpenPGP+secret+key:%0A%22testing%22%0A256-bit+ECDH+key,+ID+3ECC4691,%0Acreated+2016-05-16+(main+key+ID+317EDBEF).%0A' agent.py:172
2016-05-16 21:23:02,610 INFO       got: 'PKDECRYPT'                                                                                     agent.py:172
2016-05-16 21:23:02,610 INFO       line: 'D (7:enc-val(4:ecdh(1:s49:0x\xd1\xf4^\xf5-Q~\t\xdd`^]\x893\xa0\xe4 ^\xf6y\xd7\xdd\xb8j\xcd\xe0N\x05\x1b\x93#\xa3\xd0\x96\xe8(\x94\\}\x81H\xf8\xe6\xfdKw\x11)(1:e65:\x04o\xa0-\xae{\xd7q\xdcC=alE{9GfH\x80!\x85\xb6%0Awg\x08@m\x90\xc9\xabY\xc4x\x7f5E~U\xc2\xaa\xb1\x05:\xccZ\x17Q\xdc\x80%25R<W\xa1\x8eu\xb22CZ\xb8\x1f\x8e)))' agent.py:184
2016-05-16 21:23:02,611 INFO       exp: ['enc-val', ['ecdh', ['s', '0x\xd1\xf4^\xf5-Q~\t\xdd`^]\x893\xa0\xe4 ^\xf6y\xd7\xdd\xb8j\xcd\xe0N\x05\x1b\x93#\xa3\xd0\x96\xe8(\x94\\}\x81H\xf8\xe6\xfdKw\x11'], ['e', '\x04o\xa0-\xae{\xd7q\xdcC=alE{9GfH\x80!\x85\xb6\nwg\x08@m\x90\xc9\xabY\xc4x\x7f5E~U\xc2\xaa\xb1\x05:\xccZ\x17Q\xdc\x80%R<W\xa1\x8eu\xb22CZ\xb8\x1f\x8e']]] agent.py:188
2016-05-16 21:23:02,611 INFO       pubkey: '\x04o\xa0-\xae{\xd7q\xdcC=alE{9GfH\x80!\x85\xb6\nwg\x08@m\x90\xc9\xabY\xc4x\x7f5E~U\xc2\xaa\xb1\x05:\xccZ\x17Q\xdc\x80%R<W\xa1\x8eu\xb22CZ\xb8\x1f\x8e' agent.py:190
[... cut ...]
2016-05-16 21:23:05,620 INFO       reply: "D (5:value65:\x04k\x8b\xc6\xe4\xd8L\xec\xe2\xf6%0D\xdd\x9a\x18q{|\xc1\xb8TT\x8c\x91'\x1b%0D7\x85qEM\xf7\xa5\xbd+\xbe\xd7\xf2\xf9\x9cN\x0cf\xe9n\xc7\xca\xc4\x19\xc3v\xc6\x95cw\xea\xc6M\x8dx\xd9#\xbd\xd0\x9d)\nOK\n" agent.py:202
2016-05-16 21:23:05,620 INFO       waiting on /home/roman/.gnupg/S.gpg-agent                                                            agent.py:167

gpg: encrypted with 256-bit ECDH key, ID 3ECC4691, created 2016-05-16
      "testing"
Mon May 16 21:23:02 IDT 2016
```

jhoenicke, 2016-05-28:
First, I really like this work.

As you already noted, this should be a separate API.  Something like "ECDHSessionKey". It should also use a new subbranch of the BIP-32 hierarchy.  Maybe you can ask for a new SLIP number and add a short description to satoshilabs/slips?

Do you plan to make a Trezor-aware full gpg-agent?   That would be great.  I imagine a small server that listens on the gpg-agent socket, checks if the Trezor can be used to answer the requests, otherwise it forwards it to the old gpg-agent.  Also it should run even if no Trezor is connected and, when something should be signed with the Trezor, it should wait for it to be connected.

romanz, 2016-05-28:
Thanks, Jochen!

@prusnak May I use SLIP-17 for this feature? 
IIUC, it should be similar to SLIP-13, but instead of using `m/13'/A'/B'/C'/D'` derivation, it should use `m/17'/A'/B'/C'/D'`.
I also think that "ECDHSessionKey" is a good name for the new API. 
What is your opinion on the issue?

I agree that using a Trezor-based `gpg-agent` interface (as was done for the `ssh-agent` case) would be much better for the user experience (since it will allow using the original `gpg2` binary, while the GPG trezor-agent is running).
I will refactor the existing signing functionality to support this mode of operation, and then add the decryption functionality to use it as well.

prusnak, 2016-05-28:
SLIP17 is fine if it is not assigned yet. Also I am fine with GetECDHSessionKey/ECDHSessionKey message names.

romanz, 2016-06-17:
Thanks for the review and merges :)

romanz, 2016-06-20:
Forgot to merge the above PR (adding Python support for ECDH TREZOR API).

prusnak, 2016-06-21:
There's no need to have this reopened.

closed 2016-06-21 by prusnak