trezor/trezor-mcu #94: Core Drivers

mcccs, 2016-06-19:
Are Core Drivers (Trezor for Bitcoin Core) going to be added?
I asked Core developers and they told me that I had to ask SatoshiLabs.
> This is really something you should be asking SatoshiLabs. Core has supported the HD wallet primitives that would be needed to support hardware wallets since a few years now, they just haven't written the driver code.

Sincerely, @MCCCS

prusnak, 2016-06-19:
Sorry, but that particular Core developer is bullshitting you. BIP32 was
just added recently and BIP44 is still not ready in Core. We will not
put any effort until this is implemented by Core.

closed 2016-06-19 by mcccs

mcccs, 2016-07-31:
Also core developers don't care hardware wallet feature. Will you open a Pull Request for BIP 44?