trezor/trezor-mcu #96: Recovery without leaking parts of the seed.

jhoenicke, 2016-06-28:
The current recovery procedure leaks the words used in the seed.  I have some alternative methods unter the branches buttonrecovery/letterrecovery that implement some alternative approaches.  

I'm currently thinking of a new way that is very similar to PIN entry.  When Trezor asks for the first word it displays something like

```
 C    F-G   H-L
A-B   T-Z   M-O
P-R    S    D-E
```

The user uses the pinmatrix to choose the first letter.  Then the display switches, e.g. if `T-Z` is chosen it displays:

```
TA-TE    U    TU-TY
WA-WH  TH-TI    V
 TR    WI-Z    TO
```

The third level is similar but may contain some empty spaces:
```
TRA[C-N]   TRU
   TRE   TRA[P-Y]   TRI
   TRO              TRY
```

The fourth level only needs six choices, e.g., for `trac-tran`:

```
trade     track
transfer  traffic
tragic    train
```

I made a preliminary analysis of the english word list and four levels per word is enough with the last level only containing six choices.

jhoenicke, 2016-06-29:
Demo implementation in
 https://github.com/jhoenicke/trezor-common/tree/matrixrecovery
 https://github.com/jhoenicke/trezor-mcu/tree/matrixrecovery
 https://github.com/jhoenicke/python-trezor/tree/matrixrecovery

spudowiar, 2016-08-18:
I've been using this, works very well with `trezorctl` :+1:

prusnak, 2016-11-23:
Fixed in https://github.com/trezor/trezor-mcu/pull/130

closed 2016-11-23 by prusnak