trezor/trezor-mcu #102: U2F don't work when Trezor is loaded from XPUB

ondrejsika, 2016-08-17:
I tried U2F here https://demo.yubico.com/u2f. When I initialized Trezor classic way, it worked correctly. When I load XPUB into Trezor, after U2F request confirmation, the app frozen or failed.

Another test I've done with django-u2f test server which has similar results.

The error code is 4 (DEVICE_INELIGIBLE).

prusnak, 2016-08-17:
You cannot load XPUB into TREZOR. Do you mean XPRV using LoadMessage?

// cc @jhoenicke

ondrejsika, 2016-08-17:
Sorry, XPRV

jhoenicke, 2016-08-17:
Yes, this doesn't work.  The reason is that the xprv is only used to generate secp256k1 keys, while u2f needs a nist256 key.  It should behave as if the device isn't initialized.

Background: we now generate different "xprv" for different curves from the master seed.  This is for security reasons; we don't want that a flaw in the nist-curve leads to compromised bitcoin keys.  Load_device only allows to set one xprv though.  Since this method is inherently insecure (it exposes the xprv to the computer) and therefore deprecated, I didn't bother to setup a solution for this.

jhoenicke, 2016-08-21:
BTW, is there a reason you need to set a xprv?  I'm thinking of removing this feature completely.   You can still load_device with a bip39 seed for testing, but this way you can't use a specific xprv.

ondrejsika, 2016-08-21:
I don't know. I used the XPRV load because I was accustomed.

prusnak, 2016-08-21:
There is a valid usecase for loading XPRV when you want to have
hierarchy between the devices. E.g. one master trezor for director and
several slave trezors for branch managers.

This would need support in the wallet software, which did not happen
yet, so I am fine with removing XPRV load completely for the time being.

Links:

  1. https://github.com/trezor/trezor-mcu/issues/102#issuecomment-241254186
  2. https://github.com/

prusnak, 2016-11-09:
Closing as this is expected behaviour.

closed 2016-11-09 by prusnak