trezor/trezor-mcu #109: Support for Ethereum Classic

arvicco, 2016-08-29:
Now that Ethereum support is added to Trezor, it should be trivial to enable support for ETC as well. Ledger Nano S already has Classic support.

By enabling both ETC and ETH, you guys will be killing two birds with one stone! ;)

prusnak, 2016-08-29:
There is no difference between ETH and ETC from the TREZOR perspective, so it will already support both.

closed 2016-08-29 by prusnak

arvicco, 2016-08-29:
@prusnak Thanks for the explanation! Will there be support for both realized in Trezor webwallet interface? I understand it's not a firmware issue, so could you please point me to the appropriate place where it's being discussed? Thanks!

prusnak, 2016-08-29:
No ETH/ETC in TREZOR web wallet for now.
We'll be working with Ethereum developers to add TREZOR support to their wallets instead (such as Mist for example). It makes more sense in the long run.

arvicco, 2016-08-29:
Who should ETC developers talk to to cooperate on TREZOR support for ETC wallets?

prusnak, 2016-08-29:
With ETH developers ...

arvicco, 2016-08-30:
Why would they talk to ETH developers if we need Trezor support on ETC-only wallets? I meant, who would be a person in Trezor team in charge of such integration?

jhoenicke, 2016-08-31:
See: https://github.com/trezor/community
Maybe the best place is to ask in gitter chat (link is on the page above).
Or if you prefer a different more permanent communication format, you can also open an issue in the community repository.

arvicco, 2016-09-01:
Thank you, @jhoenicke!