trezor/trezor-mcu #112: [Hackathon] Graphene (Steem, BitShares, Muse, PeerPlays) Trezor Support

xeroc, 2016-09-01:
Hi there,

in preparations for the Hackathon in late September, I'd like to look into adding Graphene-based blockchain tech to TreZor. Since they format is quite different, I wrote a post on Steemit that [explains the whole procedure](https://steemit.com/steem/@xeroc/steem-transaction-signing-in-a-nutshell#)

Usually all transactions in Graphene are represented as JSON contructs:

``
{
      'ref_block_num': 36029,
      'ref_block_prefix': 1164960351,
      'expiration': '2016-08-08T12:24:17',
      'operations': [['vote',
                      {'author': 'xeroc',
                       'permlink': 'piston',
                       'voter': 'xeroc',
                       'weight': 10000}]],
      'extensions': [],
      'signatures': [],
}
```
Serialization would need to happen in Trezor as well as adding and providing the signature for the transaction.

* My first question would be if there is a way for Trezor to read and interpret JSON formatted transactions?
* What size limitations do we need to consider when dealing with "large" transactions?
* We also need a way to represent the date string (expiration) as an epoch-integer

Looking forward to meeting you in Prague

jhoenicke, 2016-09-01:
> My first question would be if there is a way for Trezor to read and interpret JSON formatted transactions?

No, currently there is no JSON implementation. Trezor uses protobuf for all communications with the PC.  The PC software should convert the data into a format that is understood by Trezor and collect the signatures.  In your case I would suggest, sending the serialized data to the Trezor, since that should be easier to parse on device.

> What size limitations do we need to consider when dealing with "large" transactions?

If a transaction is larger than a few kilobytes, it must be given to Trezor in small chunks that it can process in the right order.  Usually what Trezor needs to do is (1) present the information to the user to confirm, (2) check for consistency, and (3) compute a hash that is signed in the end.  Especially (3) drives the order in which the data needs to be sent to the Trezor. 

> We also need a way to represent the date string (expiration) as an epoch-integer

We don't have support for date conversion yet. Also note that the Trezor does not know the time and therefore has to rely on whatever the PC side claims is the current time.

xeroc, 2016-09-02:
> No, currently there is no JSON implementation. Trezor uses protobuf for all communications with the PC. The PC software should convert the data into a format that is understood by Trezor and collect [...]

I came to the same conclusion yesterday. Thanks for confirming!

> If a transaction is larger than a few kilobytes, it must be given to Trezor in small chunks that it can process in the right order. Usually what Trezor needs to do is (1) present the information to [...]

In the case of STEEM, people may want to sign "posting" operations with trezor. Postings can technically carry any size of information (up to some tx size limit the hundreds of kBs). However, to derive the hash for signing, trezor would need to have the WHOLE transaction in memory. Having chunks is insufficient. But I think we can still start and restrict the size of signable posting operations to whatever trezor can manage. We could even let trezor sign the hash of the transaction directly (as provided via interface), then however, trezor cannot show transaction details.

> We don't have support for date conversion yet. Also note that the Trezor does not know the time and therefore has to rely on whatever the PC side claims is the current time.

In our case, trezor doesn't need to know the time. The expiration field would be provided externally (as part of the transaction). Considering Point 1) (ProtoBuf instead of Json), I think we can do the conversion on the computer first and only let trezor deal with the timestamp as an integer.

I'll take a closer look and am hoping. Closing this ticket for now

closed 2016-09-02 by xeroc

jhoenicke, 2016-09-02:
You don't need to have the WHOLE transaction in memory, if you want to compute the sha256 hash.  The hash can be computed incrementally calling sha256_Update on each chunk.  We already use this trick for signing bitcoin transactions and large ethereum transactions.

xeroc, 2016-09-02:
> The hash can be computed incrementally

Wow. I didn't know that.