trezor/trezor-mcu #117: Update Zcash support for new address prefixes

daira, 2016-10-09:
(This was mentioned in https://github.com/trezor/trezor-mcu/pull/111#issuecomment-252083557 but that PR was already merged.)

We changed the address prefixes for transparent addresses in Zcash v1.0.0-beta-2. The new prefixes are implemented in https://github.com/zcash/zcash/pull/1477/commits/702721c1fbde449a02a91a08bf63c3d7a6a7900c ; note that they are two bytes rather than one byte. The first byte is distinct from any other coin that Trezor supports, for both Zcash mainnet and testnet.

The address prefixes used in v1.0.0-beta-2 will not change again before launch. The key format has not changed and is the same as Bitcoin; only the address format has changed. See section 5.6.1 of our [protocol spec](https://github.com/zcash/zips/blob/master/protocol/protocol.pdf) for details.

If the Zcash prefixes were one byte then I'd submit a new PR, but I'm unsure how to support a two-byte prefix. It is assumed in [transaction.c: compile_output](https://github.com/trezor/trezor-mcu/blob/master/firmware/transaction.c#L58) that address prefixes are one byte; that code looks relatively straightforward to change, but I don't know whether anywhere else also makes this assumption.

prusnak, 2016-10-09:
There are lots of places where this needs changing as I was warning in your "let's change the address prefix" issue.

trezor-crypto is already patched, but trezor-mcu needs lots of changes, these being just the tip of the iceberg: https://github.com/trezor/trezor-mcu/tree/multibyte_addrver

(fsm.c, transaction.c not finished, crypto.c not even started)

Feel free to send PRs to multibyte_addrver branch.

daira, 2016-10-09:
Oops, I didn't see your comment before sending that PR, so we might have duplicated some work (although I did see that you'd already patched trezor-crypto and made use of that). Shall I rebase my branch on top of yours?

daira, 2016-10-09:
I merged your branch into mine, since it looked like I was further along and it was simpler than rebasing. Can you have a look what things still might need changing?

daira, 2016-10-09:
OK, this branch is now ready for review (although it's entirely possible that I've missed stuff, and note this bug in trezor-crypto: https://github.com/trezor/trezor-crypto/commit/8764e26368a639f9bd34fa1c8e4378dde938589d#commitcomment-19354874 )

prusnak, 2016-10-09:
Thanks! I fixed the bug in trezor-crypto. Let's close this issue and discuss further in https://github.com/trezor/trezor-mcu/pull/118

closed 2016-10-09 by prusnak