trezor/trezor-mcu #126: Interest / Possibility of Trezor firmware written in Rust?

throughnothing, 2016-11-09:
I'm wondering if there is any interest in development of a Trezor firmware in Rust.  I've been incredibly impressed with the Trezor developers and team, and their constant dedication to the best security.

Having a firmware implemented in Rust, rather than C, could instill a lot more confidence in the correctness of the code.  Early on in Trezor development there was a BufferOverflow issue that could have been avoided, and I'm sure many other instances and classes of errors that could have been (and be avoided into the future) by avoiding incredibly error-prone languages like C.

Something like a Rust implementation could be a parallel (and even community) effort, until it would reach feature parity with this canonical implementation.  I'd definitely be interested in helping out with one, if there was enough interest and enough people who would also be willing to contribute.

prusnak, 2016-11-09:
Rust seems like an interesting language and we'll keep an eye on it, but
currently it's outside of our scope.

We will maintain TREZORv1 code still for quite some time, but we are
focusing our new development on TREZORv2 which will leverage the use of
Micropython. See trezor-core repo for more details.

saleemrashid, 2016-11-09:
I have been working on a Rust version of the firmware, and it works reasonably well. It also runs on Android (with the same codebase) if you have a kernel supporting USB HID. It is a WIP and isn't on GitHub at this point in time. @

slush0, 2016-11-09:
I'm definitely for experimenting with different software stacks and I think Rust may have sense here. So if you're Rust dev, just go for it. However as @prusnak mentioned, just now it is out of our scope as we're focusing to v2 delivery.

TREZOR v2 will allow 3rd party signed firmware, so having alternate stack would make sense for those who are more familiar with Rust or who like Rust on embedded more than Micropython.

prusnak, 2016-11-09:
Closing this issue. Interest in Rust noted, but not really related to this repository.

If you want to continue discussion, please create an issue here: https://github.com/trezor/community which is a platform for exactly this.

(If github allowed moving of the issues between repositories I would use that ... sigh)

closed 2016-11-09 by prusnak

throughnothing, 2016-11-09:
Thanks so much for the thoughts and details!  Again, I really appreciate all the hard (and great) work your team is doing.  Keep it up :)

throughnothing, 2016-11-09:
I hadn't realized the v2 firmware will be micropython.  That is interesting.  I don't know much about micropython, but I did notice that you all are using flow for your trezor.js implementation, which I think is awesome!  Makes me wonder if MyPy is a possibility to use with micropython.  I think types are pretty important for confidence and correctness in code, and is particularly important for things like crypto, and the space trezor is in.  May be something worth looking into for similar reasons that I imagine you chose to use flow with your JavaScript.

saleemrashid, 2016-11-09:
Also, see related issue trezor/python-trezor#78

prusnak, 2016-11-09:
@throughnothing Yes. Micropython is Python 3 implementation so we use type annotations that can be used with MyPy.