trezor/trezor-mcu #136: segwit support

greenaddress, 2017-01-04:
Can we have some pointer for testing segwit support with a Trezor v1 (via third party app) please?

Should we use 1.4.1 or wait for a new version? would it be in debug mode for now? and can we use it against testnet since segwit is active there?

Didn't try yet because from https://github.com/trezor/python-trezor/pull/90 it seems it is not clear and from google and repo grep didn't find much confirmation either

Thanks!

prusnak, 2017-01-04:
Segwit is not merged into the firmware (yet). I'll update this issue once it is done.

greenaddress, 2017-01-06:
thanks @prusnak

jhoenicke, 2017-01-07:
The segwit branch on trezor-mcu is mostly working now, but consider it experimental.  Don't put a valuable seed on it.  You can test segwit transactions on Testnet.  The only thing still missing from the firmware are segwit change addresses when sending transactions. 

I decided to go with BIP-142 for segwit addresses for now, even though the standard is deferred.  If later a different standard is approved, support for these addresses may be removed.  But you can use P2SH addresses instead to stay compatible to the standards.

greenaddress, 2017-03-15:
thanks @jhoenicke will test

greenaddress, 2017-03-17:
Do you know already what the firmware version number is going to be with SegWit support?

prusnak, 2017-03-17:
Good question. We'll bump the minor version - so 1.5.0

greenaddress, 2017-03-17:
thanks @prusnak !

There seems to be missing change support for segwit multisig? So it would have two prompts rather than one?

We are adding support for Trezor@SW and noticed this; for non segwit multisig you have support for change such that Trezor only prompts once

thanks!

prusnak, 2017-03-17:
> There seems to be missing change support for segwit multisig? So it would have two prompts rather than one?

Maybe @jhoenicke can answer this better.

greenaddress, 2017-03-17:
@jhoenicke I just realized you already said "The only thing still missing from the firmware are segwit change addresses when sending transactions." so this is a known limitations for now I guess. Sorry for the noise!

We'll wait for when this part is also implemented to do our testing

Thanks @prusnak

jhoenicke, 2017-03-17:
Yes, I hope to fix it in week after the next.

The reason I left this open is that one has to be careful what to allow.  You don't want to allow a malicious app to send the change to a segwit address (without any message to the user) before the soft fork activates.  The current plan is to only allow the change address to be of the same type as all inputs (similar to how multisig already works).  So a segwit change address is only allowed if all inputs are segwit.  Does this fit your use case?

greenaddress, 2017-03-17:
We do support mixed segwit and non-segwit input and so does Ledger in our testing.

In the beginning users would start with non segwit inputs, eventually have some mixed segwit and non segwit inputs and if they spend all their inputs end up with only segwit utxo at the end.

Wouldn't it be as prudent to just allow segwit output as long as at least 1 input is segwit?

Thanks

jhoenicke, 2017-03-17:
The problem is that everyone could create a dust segwit input and include it in the transaction.  You would at least need to check that the segwit change is smaller than the sum of the segwit inputs.

greenaddress, 2017-03-17:
Do you mind walking me through the scenario a bit? I'm a bit confused, are we talking post activation now?
 is the dust segwit input on chain?

greenaddress, 2017-03-17:
I'm supportive of security measures though I would have thought that the HW has no way to know if segwit activated or not and should not try to guess or prevent segwit signatures -  if we can find a clean way to avoid issues while preserving the user experience it would be great!

jhoenicke, 2017-03-17:
The concern is that pre-activation or when the user wants to opt-out of segwit, a wallet author may trick him to send the change to his segwit address without any warning on the Trezor.  For this the malicious wallet author sends some cents to his segwit address (which works pre-activation especially with p2sh).  Then he can create a transaction that includes this small segwit input and some large normal inputs and send a large change to another segwit address.  I'm not sure if segwit transactions are normally accepted by miners pre-activation, but since segwit is a soft fork a miner could include the transaction in a block.

If we check that the change contains less money than the segwit inputs, then it should be safe.

greenaddress, 2017-03-17:
It is my understanding that sw transaction pre activation don't even propagate. This would require a wallet developer and miner coordination, is it correct?

jhoenicke, 2017-03-17:
Transaction to p2sh segwit can't be distinguished from normal p2sh transactions and will propagate and confirm (there are already at least three p2sh segwit outputs on the bitcoin block chain).  So the first step would work.  

But confirming the signed segwit transaction and the transaction that steals the change output needs miner coordination, see https://blockchain.info/de/address/3DcSoJtoLYYS5yvdPG3FubWKg7kYUxzg8N, which still contains more than 3 bitcoin.

greenaddress, 2017-03-17:
I see. What about keeping it as debug only firmware until it is active at which point allow a segwit multisig change?

jhoenicke, 2017-03-29:
The way we will do it now, is to treat the change output as normal output if something is wrong with it.  So the trezor will ask whether to send x bitcoin to the address 3xy... If the inputs contains more than the change already in segwit inputs, this question will be omitted.

We also want to make the bip-32 path of the change address more restrictive while we are at it.  Do you use the public derivation path "1/<index>" for the change address?  And only use one xpub per transaction?

jhoenicke, 2017-04-02:
Note that the pc software don't have to explicitly handle the case where there is an extra confirmation of thee change output, except that it must be able to handle an extra ButtonRequest.

prusnak, 2017-05-10:
Segwit support has been just merged into master (https://github.com/trezor/trezor-mcu/commit/f1db85948ec8d7028c2b0669ffd1cc44252966b5)

Thank you Jochen for your hard work and making this possible!

closed 2017-05-10 by prusnak