trezor/trezor-mcu #142: Implement EIP 155

runn1ng, 2017-01-20:
We should implement EIP 155 for Ethereum replay attack prevention.

https://github.com/ethereum/EIPs/issues/155

When I looked into the code, I think we are using the old scheme

runn1ng, 2017-01-20:
cc @axic

runn1ng, 2017-01-20:
we should also make sure this doesn't break ETC (I am not sure how it works over there...)

jhoenicke, 2017-01-21:
I can't find if ETC has implemented eip-155 or something else (ecip-1011/1012) ... the web search is a bit confusing.

I will implement eip 155.  I just add a chain_id to the ethereum_sign_tx message.  If that is not set, a (not replay protected) signature in the old format is produced.

jhoenicke, 2017-01-22:
Note that the chain_id patch is still completely untested.

prusnak, 2017-01-22:
Implemented in https://github.com/trezor/trezor-mcu/pull/146

Please reopen if problem if issue is not fixed.

closed 2017-01-22 by prusnak

runn1ng, 2017-01-23:
OK, I will try to test this with myetherwallet.

runn1ng, 2017-01-24:
The EIP code seems to be working. The transaction has correct `v` and the signature is correct. :+1: