trezor/trezor-mcu #143: Ethereum - Wrong transaction signature with empty data/zero value

runn1ng, 2017-01-20:
If the signed transaction has empty data or zero value, the behavior is inconsistent.

For example, with empty data, when the data_length is set at 0 and data_initial_chunk is empty array, the transaction is seemingly signed and displayed on the device, but the returned signature is malformed. You need to unset data_length and data_initial_chunk.

The same with zero value (which is not uncommon with sending contracts). If the value is '0x00' (in bytes), TREZOR happily shows it correctly, but then the signature is wrong.

I think trezor should detect this and either reformat the inputs, or return an error. The current behavior - show correct data on display, but return the wrong signature.

prusnak, 2017-01-20:
cc @axic

runn1ng, 2017-01-20:
Also nonce 0x00 has the same issue

runn1ng, 2017-01-20:
....hm, but with missing nonce, I got "safety check failed" from trezor. Maybe I will just force set the nonce to "01" when it's "00", that might work? I don't know.

runn1ng, 2017-01-20:
OK, it worked not with empty (missing) nonce, but with empty byte array. However, the same trick didn't work with `data` and I had to set it as missing, not as an empty string.

So data and value has to be missing when empty, but nonce has to be empty byte array.

Luckily this can all be set in layers above trezor (python-trezor, trezor.js), but it is still confusing to anyone implementing it again.

jhoenicke, 2017-01-20:
I thought, I had tested these cases, but probably only with whatever python-trezor does for them.

As I see is the problem is that protobuf allows two ways to give no data: omit the field or pass an empty byte array.  Should we just change the firmware code that it handles these two cases identical?

prusnak, 2017-01-20:
Omitting the field and sending empty string should could stay different
operations. I have not a strong opinion whether they should be unified
in Ethereum context or not.

What I see as a problem is that sending 0 as value and 0 as nonce has to
be achieved via different way (null vs ''), which is indeed confusing.

Links:

  1. https://github.com/trezor/trezor-mcu/issues/143#issuecomment-274105367
  2. https://github.com/

jhoenicke, 2017-01-21:
I started fixing these issues.  

https://github.com/trezor/trezor-mcu/compare/master...jhoenicke:ethereum_fixes?expand=1

It's strange if you get a wrong signature for data_length=0.  I think the old code should just return an error message.  For data = "\0x00" you should get a signature but a different one than for no data (rle is 0x00 instead of 0x80).

runn1ng, 2017-01-21:
I can make a quick javascript code demonstrating the issues.

jhoenicke, 2017-01-22:
@runn1ng example code helps.  I suspect it may also be related to wrong encoding of other fields.  Note that value must be the empty array, not the "\0" array if the value is 0 (leading zeros must always be dropped).

jhoenicke, 2017-01-22:
For the case where the signature doesn't match, my suspicion is that you didn't remove leading zeros. The fields `nonce`, `gas_price`, `gas_limit`, and `value` are encoding numbers and leading zeros must be removed.  This even means giving an empty byte array if the number is 0.   If you don't remove the leading zeros, Trezor will still sign it, but you can't build a valid transaction with that signature.

On the other hand, the fields `to`, and `data` are raw objects and leading zeros must not be removed.  In particular `to` should be the empty array for contract creation and `"\0"*20` for the (unlikely) case that the public key hashes to zeros.  Do not  remove leading zeros from ethereum addresses!

prusnak, 2017-01-22:
Implemented in https://github.com/trezor/trezor-mcu/pull/146

Please reopen if problem if issue is not fixed.

closed 2017-01-22 by prusnak

runn1ng, 2017-01-23:
I made a "small" test-case.

First, you need to:

- set the trezor to the testing "all * 12" seed
- turn off the PIN

Then run this in node (with the modules installed etc) and connect that trezor:

````javascript
'use strict';

// installed from npm
var trezor = require('trezor.js-node');
var Ethereumtx = require('ethereumjs-tx');

var HD_HARDENED = 0x80000000;

// 44'/60'/0'/0 is what myetherwallet uses for 1st account == 1st address
var address_n = [ (HD_HARDENED | 44) >>> 0, (HD_HARDENED | 60) >>> 0, (HD_HARDENED | 0) >>> 0 , 0 ];
var nonce = '01';
var gas_price = '098bca5a00';
var gas_limit ='a3ff';
var to = 'dc7359317ef4cc723a3980213a013c0433a33891';
var value = '01';
var data = 'baba';

var list = new trezor.DeviceList();
var txparams = {
    "nonce": "0x" + nonce,
    "gasPrice": "0x" + gas_price,
    "gasLimit": "0x" + gas_limit,
    "to": "0x" + to,
    "value": "0x" + value,
    "data": "0x" + data,
};

list.on('connect', (device) => {
    console.log("connected device");

    device.runAggressive((session) => {
        console.log("Running call");

        return session.typedCall('EthereumSignTx', 'EthereumTxRequest', {
            address_n,
            nonce,
            gas_price,
            gas_limit,
            to,
            value,
            data_initial_chunk: data,
            data_length: data.length / 2,
        }).then((res) => {
            txparams.v = new Buffer([res.message.signature_v]);
            txparams.r = new Buffer(res.message.signature_r, 'hex');
            txparams.s = new Buffer(res.message.signature_s, 'hex');
            var tx = new Ethereumtx(txparams);

            console.log(tx.getSenderAddress().toString('hex'))
            console.log('(Should be: 1ec5c1854e3e9f1674c34d6c2be1bf13dfc0fd8f)')
        });
    }).catch((error) => {
        console.error(error);
    });
});
````

This waits for trezor and tries to sign transaction, then creates a transaction with the signature and tries to decode the sender back.

It works (with 1.4.1; haven't tested your new firmware yet, will now) with the data in the example.

However, what breaks it:
- value '00' (gets corrected by setting value to either empty string or removing fields from message)
- nonce '00' (dtto)
- data as empty string and data_length missing - that produces a bad signature - removing data from message fixes that
- data as empty string and data_length 0 - which doesn't return a bad signature, but throws a different error from trezor - removing the data_length and data_initial_chunk fixes that

runn1ng, 2017-01-23:
@jhoenicke 
> For the case where the signature doesn't match, my suspicion is that you didn't remove leading zeros. The fields nonce, gas_price, gas_limit, and value are encoding numbers and leading zeros must be [...]

Oh right, that probably is the case! I was re-sending the data from the myetherwallet representation, and they do include leading zeroes, like '0x00' for nonces.

I am not sure if "signing, but producing incorrect signature" is a correct behavior. Shouldn't the signing fail in that case?

runn1ng, 2017-01-23:
thanks to this example, I actually found a bug in the transport code :) the javascript-based transports (node hid transport, extension) differently deal with `null` in value in json->protobuf encoding when the field is optional, than bridge. :)

jhoenicke, 2017-01-23:
If you include leading zeros, then what we sign is the transaction where the leading zeros are encoded.  But maybe it makes sense to return with an error instead, since such a transaction wouldn't be legal.  Note, that ethereum_sign_tx doesn't return the raw transaction, so you don't really see the transaction that it signs.

The case `data_length` missing, `data_initial_chunk` set to empty array should have worked.  At least the two places where `has_data_initial_chunk` is used are okay.  Maybe your test had another 0 field in it.

For backwards compatibility you should still omit data_length and data_initial_chunk if data is empty and send empty array `nonce` if it is 0.  You can set `chain_id` (EIP-155) for any firmware version; it is just ignored by old firmware, which can be easily determined by the returned `v` value.

runn1ng, 2017-01-23:
> The case data_length missing, data_initial_chunk set to empty array should have worked. At least the two places where has_data_initial_chunk is used are okay. Maybe your test had another 0 field in it.

You are probably correct.

I am not really sure about the zeroes. In the end maybe you are right, let the user sign what he wants to sign. At least we now have it on github, so people will google it if they will look for it :)

I can later add these to python tests? It would make sense to test for those cases.