trezor/trezor-mcu #150: using RNG peripheral before setup?

mruddy, 2017-02-14:
I was reviewing the RNG code alongside the [reference manual](http://www.st.com/resource/en/reference_manual/cd00225773.pdf) (the latest linked to from http://www.st.com/en/microcontrollers/stm32f205re.html), and I don't see how it's not the case that either the bootloader is relying on undocumented behavior, or the reference manual has some errors.

Since I'm not sure which is the case, I'm split on if this is an issue. My device seems to work, but I don't see a guarantee that this will continue to be the case, for example, for other production runs. Below's what I see. Please let me know what you think, or if I'm missing something.

1. Section 20.4.1 says that the RNG control register resets at 0x00000000 which should mean that the TRNG peripheral is disabled. This is in line with section 20.3.1 since step 2 activates the analog part. The way I read things, until the RNGEN bit is set, the part should not be active. And at reset, the bit is documented to be cleared.
2. For the stack guard, the bootloader attempts to read from the TRNG before calling the setup routine that enables the clock and sets the RNGEN bit in the control register. See https://github.com/trezor/trezor-mcu/blob/master/bootloader/bootloader.c#L142-L143 and https://github.com/trezor/trezor-mcu/blob/master/setup.c#L43-L44. The confusing thing is that the RNG_SR_DRDY bit must be getting set in the status register for random32 to read data, weird...
3. Also, for some reason related to FIPS, after setup is called, according to section 20.3.1, "the first random number generated after setting the RNGEN bit should not be used, but saved for comparison with the next generated random number".

mruddy, 2017-02-14:
@prusnak I made some quick custom firmware that prints the ```__stack_chk_guard``` just after the bootloader turns over control and the stack guard is always zero. So, this is actually looks like a valid bug.

saleemrashid, 2017-02-14:
> My device seems to work, but I don't see a guarantee that this will continue to be the case, for example, for other production runs.

Having a stack canary set to zero isn't going to stop it from working, but it is a security issue (reduces the stack smashing protection to protection against accidental smashing rather than malicious smashing).

> 3. Also, for some reason related to FIPS, after setup is called, according to section 20.3.1, "the first random number generated after setting the RNGEN bit should not be used, but saved for comparison with the next generated random number".

This is not too much of an issue since we probably do not need the canary to be cryptographically secure.

mruddy, 2017-02-14:
1. I agree that a constant canary could be a security issue.
2. Additionally, this use on an unitialized RNG could also hang the code indefinitely if the RNG status register for some reason never set the ready bit. I think we're wandering in unspecified territory. Since it's the bootloader, and unchangeable, there is some (unknown, but probably small) probability of loading this onto a device and immediately bricking it.
3. The canary, does not need to be crypto secure, just unpredictable, but since the RNGEN bit is not set until after setup is called, the canary is not the first real use of the generator either. So, something else is, maybe key entropy or a pin generation, and that could have more of negative impact. Better to just use the recommendation in the manual and not use the first successful read after proper initialization.
3. Also, I've done more verification and still think this is an actual issue. From looking at objdumps on the elf files and testing, I think the stack guard gets loaded into SRAM at 0x20000000 by both the bootloader and firmware binaries. It took some messing around with the reset_handler code to not overwrite that before my firmware could inspect it. Instead of being all zeros, it looks like it's a fixed 0x0000FFFF. No big difference, it's still a constant.

jhoenicke, 2017-02-16:
I agree that we should check and fix this issue.  If I see it correctly, the stck_chk_guard works in standard firmware, since for this the bootloader runs setup first, correct? The new bootloader may have this problem though.  (The older bootloaders were compiled without stack check anyway).  Unfortunately it is not possible to change the bootloader, so it can't be fixed for existing devices.  But it is very unlikely that there is a stack smash attack for the bootloader.

I'll look into this more closely in a few days.

jhoenicke, 2017-02-16:
Also note that random32() cannot return 0 on the first call, since it waits until the random is different from last, which is initially 0.

mruddy, 2017-02-16:
@jhoenicke Yes to all of what you said. I've also already got a PR that just needs review and testing https://github.com/trezor/trezor-mcu/pull/151.

Also, there could be a slight bug in the way that the current firmware tries to re-init the peripheral by calling setup a second time. After the firmware calls setup, the next call to random32 could actually be used as an entropy source which goes against the manual's recommendation. My PR fixes any chance of this possibly being a bug by removing the redundant call to setup.

On your second comment, yes, I had some invalid tests that I got confused about (already in comments above and I still might have messed it up after further review [long story]). Basically, what I need to do is pull the firmware off my device and objdump it to see the address of the bootloader's stack guard to be 100% sure. Then my custom firmware can look at that address across power resets and see if it changes. In my initial tests, I overlooked that the reset_handler was being pulled in from the vendor library which overwrites SRAM when it initializes the .data and .bss sections. That's why I saw all zeros. Eh, beginner mistakes. I ordered a dev board and will be able to test these weird conditions better after it arrives.

I also have some further review of the RNG usage that I have not mentioned because I wanted to test on my dev board first. I'll mention it later if anything comes of it.

prusnak, 2017-02-16:
> Also, there could be a slight bug in the way that the current firmware tries to re-init the peripheral by calling setup a second time. 

That's incorrect. `setup() is only called` when `APPVER` is undefined that is when firmware is built to be used without bootloader. Otherwise `setupApp()` is called.

mruddy, 2017-02-16:
@prusnak ah, darn, i see. ok, thanks. i'll update that PR in a bit.

prusnak, 2017-02-17:
Commited as https://github.com/trezor/trezor-mcu/commit/c8c85424b629e4790f0745d15a1d4368046dff33

Thank you!

closed 2017-02-17 by prusnak

mruddy, 2017-02-19:
This made me laugh... I was searching for dev board alternatives and came across [this](https://github.com/karek314/trezor-diy-development-board/blob/a1f4ce277a217e9fb35ca90a0da4f3f9adc0ac81/README.md) where someone actually encountered "Function random32() caused infinite loop preventing device from starting." They thought this was an "easter egg".

karek314, 2017-02-19:
I have corrected readme in repo with boards, i haven't checked code behind that issue but taking in count Trezor developers resistance to publish firstly firmware and then bootloader, it might be true that this issue has been made on purpose in public repo to make slightly harder to create cheap copies. If this not true, please guys accept my sincere apologies.

prusnak, 2017-02-19:
@karek314 apology accepted

saleemrashid, 2017-02-19:
I can't speak for the TREZOR team, but as I said, you can verify this is the authentic code, if you have a TREZOR device, by dumping the bootloader and doing a deterministic build. Some people have written this up but I can't remember who. Hopefully, I should be able to write this up soon.

saleemrashid, 2017-02-19:
Anyway, hopefully, this has all been sorted :smile: You are definitely able to audit all the software in a TREZOR.

karek314, 2017-02-19:
That's fine, this discussion clearly proves that it was not made on purpose. I shouldn't have been speculating without deeper investigation, however i didn't care to do so.

mruddy, 2017-02-19:
@karek314 also, thanks for sharing your work! by the way, have you encountered anything else weird like this that might be worth reviewing?

karek314, 2017-02-19:
@mruddy Everything else was and is working as expected, i have just built few prototypes to work on adding support for Lisk - blockchain applications platform, although i haven't managed to find some free time to work on this, so in result i didn't worked much with code.

mruddy, 2017-02-19:
Ironically, I also just found this same bug mentioned at http://dinosaur.hiphop/ which was another dev board mentioned in the PR.

> I've also had difficulty with the bootloader getting caught in the random32 call in the beginning of main(), and the workaround is replacing it with a fixed value, for example 0x41414141.  This behavior needs to be understood better before any fulltime use.

jhoenicke, 2017-02-20:
I have checked a while ago, that the binary bootloader 1.2.0 and 1.2.5 corresponds to the source code.  I have corresponding branches in my fork of the repository, so you can see the changes (they are just reverting the changes made between the time the bootloader was released and the time the booloader source was added to the repository), e.g., https://github.com/trezor/trezor-mcu/compare/master...jhoenicke:bootloader-1.2.5

This particular bug was introduced a few months after bootloader-1.2.5 was released.

mruddy, 2017-02-20:
@jhoenicke hey could you check out https://github.com/trezor/python-trezor/pull/99. i'm having trouble reconciling the disassembly of the bootloader that i pulled from my device because the bootloader message says "Loader 1.3.0",  but the disassembly does not show the random32 function being called (which explains why it does not hang).

prusnak, 2017-02-20:
I think we built 1.3.0 w/o stack protector.

mruddy, 2017-02-20:
Ah, that would make sense then. I was just trying to figure out what is going on. Do you know if the sha256 that I got from my bootloader matches what it's supposed to be?

Also, I got a STM32 [nucleo dev board](http://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-eval-tools/stm32-mcu-eval-tools/stm32-mcu-nucleo/nucleo-f207zg.html) and was able to reproduce the hang. That's why I started looking deeper.

prusnak, 2017-02-20:
Yes, i confirm that Bootloader 1.3.0 has the following hash:

`add28d5faf6de00525bd37d221b195f6ba51e245923e1eadd3db0b7fd9d9caf5`

mruddy, 2017-02-20:
Cool, thanks!