trezor/trezor-mcu #152: Adding Lisk support

karek314, 2017-02-20:
Adding support for Lisk - blockchain application platform on mytrezor.com
It would be great to see Lisk integration on mytrezor.com with functionalities like
-Sending/receiving funds
-Voting for delegates

It shouldn't be much work, Lisk api is relatively simple and clear, voting is very similar to typical transactions.

https://lisk.io
https://github.com/LiskHQ/lisk

prusnak, 2017-02-20:
It makes more sense to integrate TREZOR into official Lisk wallet (if there is any).

TREZOR Wallet (aka myTREZOR) is only for coins that use Bitcore (or its fork) as backend.

closed 2017-02-20 by prusnak

karek314, 2017-02-20:
As far i remember, integrating any coin needs some work in firmware - ex.
https://github.com/trezor/trezor-mcu/blob/f8ad9fc74201c4c84de94a4dd9831dc54cd7cce9/firmware/coins.c

However i'm not sure.

prusnak, 2017-02-20:
This table is used for Bitcoin-based altcoins. Integrating other coins require much more work.

karek314, 2017-02-20:
Is it possible that Lisk will be added at firmware level which will allow further integration?
Are there any factors to fulfil to make that happen ?

I believe that PR for this would be the best solution, however i don't have time for this and i do not work for Lisk Foundation, not sure if Lisk Foundation have interest in that either way i believe it would be easiest and fastest to write code for those who knows this code deeply.

Let me know your thoughts.

prusnak, 2017-02-20:
If there is a pull request that meets quality standards we will merge it.

karek314, 2017-02-20:
So there is no possibility that it will be made by you/your team unless reaching popularity of Ethereum for exmaple ? Or their foundation paid for that ? Can you clarify ?

prusnak, 2017-02-20:
At the moment, we don't have capacity for doing that in-house. Foundation should look somewhere among Lisk people to do that.

Phoenix1969, 2017-02-20:
So a translation layer to conform to Bitcore is needed to add Lisk to Trezor?

prusnak, 2017-02-20:
No. It is needed just for addition to TREZOR Web Wallet.

TREZOR (hardware) can support Lisk if some other wallet integrates TREZOR.