trezor/trezor-mcu #155: rng code review

mruddy, 2017-02-21:
I came across this while reviewing and researching the RNG code.

First, I'm not sure that it applies to the Trezor hardware because I found this info in a manual for another MCU. But, the description of the hardware implementation of the RNG sounded close enough that it might apply.

Second, the threat might not be within the scope of threats that the Trezor was designed to withstand.

I figured that since I did the code review, I might as well put it out there and see. I'm not an expert in this, but maybe someone that knows more can say.

[Section 37.1.1](http://cache.freescale.com/files/microcontrollers/doc/ref_manual/K70P256M150SF3RM.pdf) of this other MCU's manual says:

"In fact, there may be an attack against this random number generator if its output is used directly in a cryptographic application. The attack is based on the linearity of the internal shift registers. Therefore, it is highly recommended that this random data produced by this module be used as an entropy source to provide an input seed to a NIST-approved pseudo-random-number generator...".

And:

"The requirement is to maximize the entropy of this input seed. In order to do this, when data is extracted from RNGA as quickly as the hardware allows, there are about one or two bits of added entropy per 32-bit word. Any single bit of that word contains that entropy. Therefore, when used as an entropy source, a random number should be generated for each bit of entropy required, and the least significant bit (any bit would be equivalent) of each word retained. The remainder of each random number should then be discarded. Used this way, even with full knowledge of the internal state of RNGA and all prior random numbers, an attacker is not able to predict the values of the extracted bits."

I'm wondering if ```random_buffer``` should be updated to behave in this manner? I could not find in the STM32F2 documentation any guarantee that each read actually received 32 bits of entropy vs 32 bits of random looking data with only a couple bits of entropy. Although, it may be implied.

The attack that I'm thinking about would be during device reset in the case where the external entropy is purposefully weakened.
The hashing together of external and internal entropy mitigates a lot. But, if the same attacker can predict the internally generated entropy too, based on the linearity described above, then the change might be worth making. I don't know how really an attacker would pull this off. Maybe via a GetEntropy request first from the device?

mruddy, 2017-02-21:
Actually, another takeaway from this might be to hash the data from ```random_buffer``` before sending it back in ```fsm_msgGetEntropy```.

saleemrashid, 2017-02-21:
@mruddy Would it then not be an equally good idea to hash the data within `random_buffer`?

mruddy, 2017-02-21:
@saleemrashid Hmm, just before returning from ```random_buffer```, probably. Although, there's still the question of how many bits of entropy does each random32 call actually provide when called in a tight loop? Hopefully it's all 32 and not 2 like the other MCU. That would be the reason for why it might be prudent to call random32 for each bit of desired data.

saleemrashid, 2017-02-21:
@mruddy For some reason, I thought `random32` used `random_buffer`, so I meant putting the hash call into `random32` perhaps. Although that would be too inefficient, most likely. I'd have to benchmark with different hash algorithms.

saleemrashid, 2017-02-21:
Although we could just put a PRNG in front of the TRNG, although maybe not a "NIST approved" one :) https://en.wikipedia.org/wiki/Dual_EC_DRBG

mruddy, 2017-02-21:
Sure, I can see where putting it in random32 could make sense from the perspective of trying not to leak raw data from the RNG. But like you say, it could be a lot of overhead especially since hashing would not be adding entropy. I kind of like your idea of hashing at the end of ```random_buffer``` though and just using that as the standard API and only using random32 carefully for internal use only / performance.

mruddy, 2017-03-01:
Interesting update on this: the idea from my original post about using only the least significant bit (or another single bit) from the random32 call is a terrible idea for this hardware in practice. I used the new entropy getter tool to grab some data from my Trezor that was loaded with a firmware that called random32 8 times for each requested byte (storing the least significant bit each time) and ran the statistical tests on the data, and it absolutely fails all FIPS 140-2 tests and has chi square values less than 0.01 percent. It absolutely is a horrible result. So, forget that idea. I still wonder whether we should hash or HMAC the data before returning it from GetEntropy just to add extra safety margin. I'll do some more testing.

mruddy, 2017-03-01:
I deleted my last comment about the least significant bit idea because I found that I had borked my test. I fixed my test and it was nowhere near as bad. Using only one bit per random32 call may allow longer runs to occur in the final data that the TRNG fault checker prevents at the 32-bit granularity. That came up in the stats for one test. It's hard to say if the LSB technique is worse than just using all 32-bits from the TRNG. It doesn't appear to be obviously better.

I'm closing this because after looking at the statistical data on megabytes of generated entropy, there doesn't seem to be a degradation in entropy when the TRNG is queried repeatedly in quick succession. The other envisioned attacks seem sufficiently mitigated. Using a non-compromised DRBG might be a good idea. But, it's not entirely clear to me that it's a priority right now with the current mitigations in place.

closed 2017-03-01 by mruddy