trezor/trezor-mcu #164: Docker build fails

runn1ng, 2017-05-08:
This happens when I try docker build

````
../common/i2c_common_v2.c: In function 'i2c_transfer7':
../common/i2c_common_v2.c:431:3: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
   for (size_t i = 0; i < rn; i++) {
   ^
../common/i2c_common_v2.c:431:3: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code
../../Makefile.include:44: recipe for target 'i2c_common_v2.o' failed
make[1]: *** [i2c_common_v2.o] Error 1
Makefile:63: recipe for target 'lib/stm32/f0' failed
make: *** [lib/stm32/f0] Error 2
make: Leaving directory '/trezor-mcu/vendor/libopencm3'
````

runn1ng, 2017-05-08:
I would add docker building to travis, but the docker script downloads the source from github, which wouldn't of course work without some rewrites

prusnak, 2017-05-08:
Fixed in 31a09a9f7501f4572a93a9d054e377d021cbe60a

PS: Let's not bloat Travis build with Docker stuff ...

closed 2017-05-08 by prusnak