trezor/trezor-mcu #165: bootloader-docker-build.sh fails on "No rule to make target ... libopencm3_stm32f2.a', needed by 'bootloader.elf'.  Stop."

fivepiece, 2017-05-14:
Running the firmware and bootloader build scripts on `v1.4.2` , firmware build passes and verifies as the same hash, but bootloader fails to complete.
On current master at `c8dc17341c5a103bbc8d5b425214e8c01d72f730` , bootloader build completes successfully.

Log of the failed build:
https://gist.github.com/fivepiece/46eaf2f0a377396a3afc50eb6ad3d6d7

Thanks!

prusnak, 2017-05-15:
Tag `v1.4.2` means that firmware v1.4.2 was built from that commit. Unfortunately bootloader build was broken in that particular commit. 

I just added tags `bl1.3.0` and `bl1.3.1`which should mark commits from which bootloader version 1.3.0 and 1.3.1 were built respectively.

closed 2017-05-15 by prusnak

fivepiece, 2017-05-15:
Thanks @prusnak .  I just tried these two new tags but am getting the same error:
`make: *** No rule to make target '/trezor-mcu/vendor/libopencm3/lib/libopencm3_stm32f2.a', needed by 'bootloader.elf'.  Stop.
make: Leaving directory '/trezor-mcu/bootloader`
on both.

prusnak, 2017-05-15:
You should use `./bootloader-docker-build.sh bl1.3.1` to build this particular tag. The script tries to build master if you don't provide a tag using command line. It does not build sources in work directory.

fivepiece, 2017-05-15:
Pebkac.
Thanks for taking the time!

prusnak, 2017-05-15:
The whole Docker build thingie is not optimal and I'd like to switch to Gitian - see https://github.com/trezor/trezor-mcu/issues/66

Docker guarantees reproducible builds at the day of the release, but everything afterwards could be moving which breaks the hash. I hope that Gitian is better in that matter, but I am afraid it is not possible to build two year old release of Bitcoin unless someone caches all dependencies forever. And then it is an issue of how much do you trust this cache.

fivepiece, 2017-05-15:
I'll be happy to test gitian builds once I figure out why `trezorctl` doesn't see my device while in bootloader mode :)