trezor/trezor-mcu #166: Trezor locks up on waking up

cryptosig, 2017-05-23:
I sometimes have the Trezor lockup during a wake up event. This happens on many different client codes, ie MyTrezor, Electrum, My phone, Password Manager, myEtherWallet. I have experienced this bug on all these platforms, the only solution is to reconnect the device. The Trezor never reestablishes connection on its own. Perhaps if the bug cannot be found, maybe a timeout and reconnection would work?

prusnak, 2017-05-23:
Have never seen a report like this in the wild. What firmware version are you using? Also it might be a faulty device.

closed 2017-05-23 by prusnak

asifmoo, 2017-07-11:
I have just received device few days back and has all the issues mentioned by crptosig from day one. Mine one even got stuck while installing firmware for the first time?

prusnak, 2017-07-11:
Go to https://trezor.io/support/