trezor/trezor-mcu #168: U2F test suite not passing

prusnak, 2017-05-29:
U2F test suite: https://github.com/google/u2f-ref-code/tree/master/u2f-tests/HID

./HIDTest passes
./U2FTest fails

jhoenicke, 2017-05-30:
I actually had some small changes in my test suite to get it through:

U2FTest uses a fresh random challenge on every request and even a fresh appid.  But Trezor actually checks that the request didn't change after the user confirmed the request.  I consider this a bug in U2FTest and patched my version of U2FTest to use the same nonce  and appid for both requests.

I tried to rerun the tests.  U2FTest passes with the change to not change the challenge for each call. HIDTest now fails for me in test_Descriptor:

    ioctl(9134128, HIDIOCGRDESCSIZE or HIDIOCGVERSION, 0x7ffd82033b64) = -1 EBADF (Bad file descriptor)

Looks like this test is no longer compatible with hidapi, as it relies on some internals that changed.  The file descriptor is retrieved by an ugly cast:

    // hidapi hides internal struct.
    // Use inside knowledge to cast and get fd we need.
    int fd = *(int*)(device->dev);

prusnak, 2017-05-30:
For me, the HIDTest works. Can I take a look at your patched version of U2FTest?

saleemrashid, 2017-05-30:
@prusnak With [this patch](https://github.com/trezor/trezor-mcu/files/1037854/u2ftest-trezor.txt), it works

saleemrashid, 2017-05-30:
```sh
$ git rev-parse HEAD
ebad530a36783b09e6a4bb3ff3765feca9f97003
$ git -C core rev-parse HEAD
8032f3b5a501dabc1e28925cca834d2a39efbdee
$ git -C hidapi rev-parse HEAD
a6a622ffb680c55da0de787ff93b80280498330f
$ ./list
Device Found
  VID PID:      534c 0001
  Page/Usage:   0x3a32/0x2e31 (14898/11825)

  Manufacturer: SatoshiLabs
  Product:      TREZOR
  Device path:  /dev/hidraw0

Device Found
  VID PID:      534c 0001
  Page/Usage:   0x3a32/0x2e31 (14898/11825)

  Manufacturer: SatoshiLabs
  Product:      TREZOR
  Device path:  /dev/hidraw1

$ ./HIDTest /dev/hidraw1
PASS(test_Idle())
PASS(test_Init())
PASS(test_BasicInit())
PASS(test_Unknown(U2FHID_SYNC))
PASS(test_InitOnNonBroadcastEchoesCID())
PASS(test_InitUnderLock())
PASS(test_InitSelfAborts())
PASS(test_InitOther())
PASS(test_OptionalWink())
PASS(test_Lock())
PASS(test_Echo())
PASS(test_LongEcho())
PASS(test_Timeout())
PASS(test_WrongSeq())
PASS(test_NotCont())
PASS(test_NotFirst())
PASS(test_Limits())
PASS(test_Busy())
PASS(test_LeadingZero())
PASS(test_Idle(2.0))
PASS(test_NothingOnChannel0())
PASS(test_OnlyInitOnBroadcast())
PASS(test_Descriptor())
$ ./U2FTest /dev/hidraw1
PASS(check_Compilation())
PASS(test_Version())
PASS(test_UnknownINS())
PASS(test_WrongLength_U2F_VERSION())
PASS(test_WrongLength_U2F_REGISTER())
PASS(test_BadCLA())
PASS(test_Enroll(0x6985))

Touch device and hit enter..

PASS(test_Enroll(0x9000))
PASS(test_Sign(0x6985))
PASS(test_Sign(0x6985, true))
PASS(test_Sign(0x6a80))
PASS(test_Sign(0x6a80))

Touch device and hit enter..

PASS(test_Sign(0x6985, true))
PASS(ctr1 = test_Sign(0x9000))
PASS(test_Sign(0x6985))

Touch device and hit enter..

PASS(ctr2 = test_Sign(0x9000))
```

saleemrashid, 2017-05-30:
This is relevant https://github.com/google/u2f-ref-code/issues/128

prusnak, 2017-05-30:
Thanks saleem, I confirm that it works. Closing!

prusnak, 2017-05-30:
Commited here https://github.com/trezor/u2f-ref-code/commit/1912ca0a3fcdaffae93dec5b8e0c57f9eb5651be and sent pull request here: https://github.com/google/u2f-ref-code/pull/159

closed 2017-05-30 by prusnak