trezor/trezor-mcu #170: Docs: How do I load the firmware using a command line interface?

jcalfee, 2017-06-09:
Can you add some tips or direction for loading the firmware?

I am pleased that I could build my own version and get a matching fingerprint.  I'm lost trying to figure out how to load it though.

prusnak, 2017-06-09:
Install python-trezor using `pip install trezor`. Then use `trezorctl firmware_update -f firwmare.bin`.

closed 2017-06-09 by prusnak

jcalfee, 2017-06-09:
Thanks .. I installed and confirmed the correct fingerprint for v1.4.2 (see below).  The device said I could unplug.  Immediately after unplugging and plugging in again it reports "unofficial firmware" and gives me a different fingerprint: 907D5921D...

```bash
trezorctl firmware_update -f output/trezor-v1.4.2.bin
Please confirm action on device...
Firmware fingerprint: a4b39f01bd134d01d7534821445bf779dbe6c25f0fcf7c7cb285a79b17f25e0a
True
```

jcalfee, 2017-06-09:
Since I relied that the device confirms the firmware fingerprint I updated the firmware using the chrome extension and did not experience the problem.  I can't be sure but it may have taken a bit longer to load.

I guess in summary two things would have made it easier:
- knowing the device confirms the fingerprint (making the chrome extension's update verifiable)
- a blurb in the readme here about trezorctl

If you want more details from me about the issue above please let me know..

jcalfee, 2017-06-09:
Wait: I get it .. My build was not signed as an official build.  That is why the fingerprint skips the first 257 bytes.  Very interesting..

prusnak, 2017-06-09:
Feel free to send pull request which improves the documentation so the situation is more obvious.