trezor/trezor-mcu #171: List of supported coins missing ethereum

almindor, 2017-06-09:
As far as I know ethereum is supported but is not listed in: https://github.com/trezor/trezor-mcu/blob/master/firmware/coins.c#L28

My device also doesn't list it in the features reply

prusnak, 2017-06-09:
Yes, the list contains only coins that are derived from Bitcoin and use
same messages as Bitcoin.
Ethereum OTOH uses different messages and thus does not need to be
listed there.
If you do think we should mention that TREZOR also supports Ethereum,
Ethereum Classic, Rootstock and all ERC-20 tokens in that particular
place in the source code, please, send a pull request with this change.
Thanks!

almindor, 2017-06-12:
I suppose it makes sense. It was just confusing there was a link in the docs somewhere (where it says ethereum is supported) pointing to this source code where it wasn't listed :)

Closing the issue

closed 2017-06-12 by almindor