trezor/trezor-mcu #181: EstimateTxSize doesn't account for segwit

runn1ng, 2017-06-28:
EstimateTxSize call doesn't account for segwit, which has different transaction sizes.

Not sure if this call is used in any software though, so maybe it's not important enough to fix

runn1ng, 2017-06-28:
In the proto file, I see "This behaves exactly like SignTx, which means that it can ask using TxRequest"; however, the code doesn't seem to do that.

jhoenicke, 2017-06-28:
> This behaves exactly like SignTx, which means that it can ask using TxRequest

Not sure if it ever did, but it doesn't do that currently.  Does it make sense to implement this?  One would need to ask for all inputs (to distinguish between segwit, multisig, etc) and outputs (op_return, or p2wsh).  Also it should probably return weight/4 instead of size, i.e. account for the segwit discount.  The question is does it help more than it breaks existing wallets that don't expect a TxRequest.

While we are there, the signing code also uses the very coarse estimate.  This may lead to fee warnings if the fee needs to be higher for multisig.  At this point Trezor could have used a much better estimate, since it already saw all inputs and outputs.

runn1ng, 2017-06-28:
> The question is: does it help more than it breaks existing wallets that don't expect a TxRequest.

Again, I am not sure if any existing wallet uses this call at all.

prusnak, 2017-06-28:
I would rather not complicate things. Let's add extra field `segwit_inputs_count` to `EstimateTxSize` message if we really need this or drop the usage of EstimateTxSize message completely.

jhoenicke, 2017-06-28:
I'd prefer to drop the support.  You don't need a secure hardware device to compute `10 + 149*a + 90*b+ 35*c`.

prusnak, 2017-06-28:
I am not aware of any wallet that is using it (I checked Electrum and our Wallet).

I will remove it from the firmware.

prusnak, 2017-06-28:
Gone in https://github.com/trezor/trezor-mcu/commit/91a1b6c4bcaa33de37ea91769028e78bf2816805

closed 2017-06-28 by prusnak

runn1ng, 2017-07-01:
Should the message be also removed from messages.proto in trezor-common?

prusnak, 2017-07-01:
No, we just mark it as obsolete via `deprecated = true` in this list:

https://github.com/trezor/trezor-common/blob/master/protob/messages.proto#L54

Maybe we can add `option deprecated = true;` to message definition as well, but I don't think there is a strong need for that.