trezor/trezor-mcu #183: Incompatibility in Firmware 1.5.0 with mycelium (and maybe other wallets).

jhoenicke, 2017-06-30:
See https://www.reddit.com/r/TREZOR/comments/6kbiyv/error_address_in_change_output_on_mycelium/
and https://github.com/mycelium-com/wallet/issues/366

The new firmware introduced a check that `address` and `address_n` (change output) should never be given at the same time in an output of a transaction.  The reasoning behind this was that the address could be wrong and if it would be used by some later code it would be able to redirect the change.

Unfortunately, this is incompatible with the way Mycelium used it.

Possible fixes

* Instead of returning an error, just clear the address field.
* Check that address matches address_n and allow if they do match.

The first is simpler and I think closer to the previous behavior.

prusnak, 2017-06-30:
Let's fix the broken software (Mycelium) and keep firmware as is.
Mycelium can update all their userbase very easily via Google Play.

prusnak, 2017-06-30:
Fixed in https://github.com/mycelium-com/wallet/pull/367

Electrum should be fine as this uses `python-trezor` internally

closed 2017-06-30 by prusnak