trezor/trezor-mcu #187: Please confirm: the IV is 16 bytes or it is not used

jcalfee, 2017-07-02:
https://github.com/trezor/trezor-mcu/blob/3c0a9c2a87c0ebea9b08908b506adfad3b9086f3/firmware/fsm.c#L545

Also, `msg->iv.size == 16` is comparing a binary size not hex right?

jcalfee, 2017-07-02:
Seems like this is a good case for check_param..

prusnak, 2017-07-02:
Yes, TREZOR does not use hex for binary data. So the check is checking
whether IV is 16 bytes or 32 hexa digits.
Also I confirm that IV is only used only if it is of correct size.

closed 2017-07-02 by jcalfee

prusnak, 2017-07-02:
Why were you asking?

jcalfee, 2017-07-03:
I was concerned that encrypting the same data in the same way more than once would result in the same encrypted data.  Kind of an edge case but it is good to be thorough when it comes to this stuff.  So I was planing to provide a random IV.  Since it looks like I would not know if I got the length incorrect I wanted to be sure.