trezor/trezor-mcu #188: Add Fujicoin support

fujicoin, 2017-07-06:
I'd like to add Fujicoin (FJC) to trezor's firmware.
I am developing Electrum-FJC now.
Https://bitcointalk.org/index.php?topic=644601.msg19958478#msg19958478

Electrum - FJC is currently being tested, but it is moving quite lightly.
When do you plan the next firmware update?

The main parameters of Fujicoin are as follows.
Do you have any missing information?

ADDRTYPE_P2PKH = 36
ADDRTYPE_P2SH = 16
ADDRTYPE_P2WPKH = None
XPRV_HEADER = 0x0488ade4
XPUB_HEADER = 0x0488b21e
GENESIS = "adb6d9cfd74075e7f91608add4bd2a2ea636f70856183086842667a1597714a0"
BIP-0044 Registered coin types: 75 | 0x8000004b | FujiCoin
fujicoin/electrum: https://github.com/fujicoin/electrum

Thanks,
motty

fujicoin, 2017-07-11:
COINS_COUNT of Fujicoin is as follows.
Firmware / coins.c:
```
const CoinType coins[COINS_COUNT] = {

	{true, "FujiCoin",      true, " FJC",  true,   36, true,   50000000, true,   16, true, "\x19" "Fujicoin Signed Message:\n", true, 0x0488b21e, true, 0x0488ade4, true, true,  },

};
```
For electrum-FJC, TREZOR correspondence has already been completed, and it is possible to test if the firmware is updated.

The repository of github is as follows.
https://github.com/fujicoin/electrum-fjc

Windows.exe is available from the following URL.
https://bitcointalk.org/index.php?topic=644601.msg20043948#msg20043948

prusnak, 2017-11-13:
Please send a pull request to trezor-common, where you would add your coin to coins.json: https://github.com/trezor/trezor-common/blob/master/coins.json

closed 2017-11-13 by prusnak

fujicoin, 2017-11-13:
Hi prusnak,

Thank you.
I have created a new pull request.

Best regards,
motty