trezor/trezor-mcu #203: Remove the arbitrary 50-character limit on passphrases

stevenroose, 2017-08-01:
Considering that the passphrase is used to generate wallets using the BIP39 spec, it is possible to have setup a valid BIP39 wallet that is not compatible with Trezor. The spec does not mention any limit on the passphrase, so any wallet with a passphrase longer than 50 characters will not be able to be imported in Trezor.

50 characters is a low limit. It is becoming more and more common to use a hash function as part of your password generation strategy. Standard SHA256 hashes, represented in hexadeximal, which is the most common representation, are 64 characters long (32 bytes). Thus, passphrases generated using a SHA256 has are not supported by Trezor.

It makes sense to have a certain limit, to prevent someone trying to break the Trezor by providing extraordinarily huge passphrases. 50 characters is very small, though.

prusnak, 2017-08-01:
I am not able to increase the limit without major rewrite of the code, which will not happen. (The reason is that the whole passphrase needs to fit into one USB response which is 64 bytes and some bytes are already used by our protocol).

closed 2017-08-01 by prusnak

saleemrashid, 2017-08-01:
@prusnak The firmware should be able to handle packets up to 12 KiB, unless I'm mistaken

prusnak, 2017-08-01:
Passphrase is handled via tiny mode, which is limited to one HID packet.

saleemrashid, 2017-08-01:
Ah, that makes sense. I don't think it will be that difficult to handle multiple packets in tiny mode. I can take a look at it later.

prusnak, 2017-08-01:
Not really important IMHO.

stevenroose, 2017-08-01:
> Passphrase is handled via tiny mode, which is limited to one HID packet. (Ugly, I know)

Maybe an opportunity to make it less *ugly* :) I, for one, would appreciate. But of course not high priority, this is not a security issue.