trezor/trezor-mcu #204: Trezor pairing with Electrum reports "read error" on OSX

ManfredKarrer, 2017-08-06:
Not sure if issue is on Electrum side or Trezors side but in doubt I post here as well (Electrum issue: https://github.com/spesmilo/electrum/issues/2698).

I am using latest Trezor firmware (1.5.0 ) and latest Electrum software (2.9.2) for OSX 10.11.6.
When I setup a new wallet with Trezor it recognize Trezor but then reports a "read error".
Tried both without Trezor wallet page open and Pin entered and without Trezor wallet page open and Pin not entered.

prusnak, 2017-08-06:
Make sure you don't have trezor bridge (aka trezord) running nor you have trezor web wallet opened in the browser when you use Electrum. Best is to close browser completely to be sure.

closed 2017-08-06 by prusnak

ManfredKarrer, 2017-08-06:
Update: Seems it works now after restarted all. Yes was probably a conflict as Trezor was used before in the default Trezor wallet.