trezor/trezor-mcu #205: SHA256 checksums of downloaded and self-built firmwares are different

serge-name, 2017-08-08:
Hello,

First, check the signed firmware:

```
% wget -q https://wallet.trezor.io/data/firmware/trezor-1.5.0.bin
% ./trezor-mcu/firmware-fingerprint.sh trezor-1.5.0.bin
---------------------
Firmware fingerprint:
c4eddafd29b580d8482cda68e61bdcf1740d77520ef3a603758646bbffe957ea  -
% ls -l trezor-1.5.0.bin 
-rw------- 1 snp snp 340620 Jun 26 15:58 trezor-1.5.0.bin
```

Second, build the firmware from sources:
```
[... skipped, full log attached ...]
Filename    : build/trezor-v1.5.0.bin
Fingerprint : a42386ddc55d4caa75c256912c176250af81562b7982108304e3735a8911ad13
Size        : 341924 bytes (out of 491520 maximum)
```
[build-1.5.0.zip](https://github.com/trezor/trezor-mcu/files/1207908/build-1.5.0.zip)

Expected result: fingerprints should be same.

prusnak, 2017-08-08:
Have you checked out the correct tag and then run the build script?

that means:
```
git checkout v1.5.0
./firmware-docker-build.sh v1.5.0
```

serge-name, 2017-08-08:
> I suspect you are using script from master to build older firmware.
That's right. My fault.

```
% git checkout v1.5.0
% ./firmware-docker-build.sh v1.5.0

[... skipped ...]

Firmware fingerprint:
c4eddafd29b580d8482cda68e61bdcf1740d77520ef3a603758646bbffe957ea  -
Firmware size: 340620 bytes (out of 491520 maximum)
```

Everything is fine now. Thanks a lot.

closed 2017-08-08 by serge-name

prusnak, 2017-08-08:
No worries. It is not very obvious, I made the same mistake too a few times.

If you have any ideas how to make this process more easy (i.e. less error prone), please share or even better create a pull request so less people will make the same mistakes in the future.