trezor/trezor-mcu #206: Firmware build fails on Fedora 26

hegjon, 2017-08-10:
Git dd9f7cd926245c5471f14473c36f7d0c1ff95f55 (master)
Docker version 1.13.1, build 27e468e/1.13.1

Output:
```
$ ./build-firmware.sh v1.5.1
Sending build context to Docker daemon 2.423 MB
Step 1/3 : FROM debian:9
 ---> a20fd0d59cf1
Step 2/3 : RUN apt-get update
 ---> Using cache
 ---> cdd7ffdf5051
Step 3/3 : RUN apt-get install -y build-essential git python python-ecdsa gcc-arm-none-eabi
 ---> Using cache
 ---> 03314fb45a90
Successfully built 03314fb45a90
Cloning into 'trezor-mcu'...
remote: Counting objects: 3404, done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 3404 (delta 0), reused 1 (delta 0), pack-reused 3400
Receiving objects: 100% (3404/3404), 1.36 MiB | 245.00 KiB/s, done.
Resolving deltas: 100% (2509/2509), done.
...
3/include -I/trezor-mcu/ -I/trezor-mcu/gen -I/trezor-mcu/vendor/trezor-crypto -I/trezor-mcu/vendor/trezor-crypto/aes -I/trezor-mcu/vendor/trezor-crypto/ed25519-donna -I/trezor-mcu/vendor/trezor-qrenc -DAPPVER=1.0.0 -DMEMORY_PROTECT=1 -DDEBUG_RNG=0 -DFASTFLASH=0 -Wno-sequence-point -Iprotob -DPB_FIELD_16BIT=1 -DQR_MAX_VERSION=0 -DDEBUG_LINK=0 -DDEBUG_LOG=0 -DSCM_REVISION='"\xf0\xd2\xe7\xa3\x71\x42\xa6\xd4\xc7\xf7\xe4\x5a\x6e\x44\x27\xe5\x31\x23\xd6\x14"' -DUSE_ETHEREUM=1 -MMD -o protob/types.pb.o -c protob/types.pb.c
arm-none-eabi-gcc -o trezor.elf usb.o u2f.o messages.o storage.o trezor.o pinmatrix.o fsm.o coins.o transaction.o protect.o layout2.o recovery.o reset.o signing.o crypto.o ethereum.o ethereum_tokens.o debug.o ../vendor/trezor-crypto/address.o ../vendor/trezor-crypto/bignum.o ../vendor/trezor-crypto/ecdsa.o ../vendor/trezor-crypto/curves.o ../vendor/trezor-crypto/secp256k1.o ../vendor/trezor-crypto/nist256p1.o ../vendor/trezor-crypto/ed25519-donna/ed25519.o ../vendor/trezor-crypto/ed25519-donna/ed25519-donna-basepoint-table.o ../vendor/trezor-crypto/ed25519-donna/ed25519-sha3.o ../vendor/trezor-crypto/ed25519-donna/ed25519-keccak.o ../vendor/trezor-crypto/hmac.o ../vendor/trezor-crypto/bip32.o ../vendor/trezor-crypto/bip39.o ../vendor/trezor-crypto/pbkdf2.o ../vendor/trezor-crypto/base32.o ../vendor/trezor-crypto/base58.o ../vendor/trezor-crypto/ripemd160.o ../vendor/trezor-crypto/sha2.o ../vendor/trezor-crypto/sha3.o ../vendor/trezor-crypto/aes/aescrypt.o ../vendor/trezor-crypto/aes/aeskey.o ../vendor/trezor-crypto/aes/aestab.o ../vendor/trezor-crypto/aes/aes_modes.o ../vendor/trezor-qrenc/qr_encode.o protob/pb_decode.o protob/pb_encode.o protob/messages.pb.o protob/storage.pb.o protob/types.pb.o -ltrezor -lopencm3_stm32f2 --static -Wl,--start-group -lc -lgcc -lnosys -Wl,--end-group -L/trezor-mcu/ -L/trezor-mcu/vendor/libopencm3/lib -L/trezor-mcu/vendor/libopencm3/lib/stm32/f2 -T/trezor-mcu//memory_app_1.0.0.ld -nostartfiles -Wl,--gc-sections -mcpu=cortex-m3 -mthumb -msoft-float
arm-none-eabi-objcopy -Obinary trezor.elf trezor.bin
make: Leaving directory '/trezor-mcu/firmware'
make: Entering directory '/trezor-mcu/firmware'
../bootloader/firmware_sign.py -f trezor.bin
Metadata has been added...
Firmware size 393412 bytes
Firmware fingerprint: 1c1fa9802cbd6a947a4f3e78f209d3efe49eb4abbacb101bbc3a0a709c742707
Slot #1 is empty
Slot #2 is empty
Slot #3 is empty
make: Leaving directory '/trezor-mcu/firmware'
cp: cannot create regular file '/build/trezor-v1.5.1.bin': Permission denied
```

prusnak, 2017-08-10:
Weird, docker should have created `$(pwd)/build` directory

hegjon, 2017-08-10:
I see that it have been created, but the owner is root and not my user. I run docker as my user

prusnak, 2017-08-10:
Huh, dunno how docker is packaged in Fedora, so I cannot comment on it (I run docker as root).

`build` directory is created by docker, so I wonder why the owner is root not your user.

hegjon, 2017-08-10:
I tried to build again, but this time I ran chown first, but I get the same error

hegjon, 2017-08-10:
Running the ./build-firmware.sh as sudo gives also the same error

prusnak, 2017-08-10:
Try removing the directory first.

hegjon, 2017-08-10:
Output from audit.log when the command fails:
```
type=AVC msg=audit(1502369722.597:748): avc:  denied  { write } for  pid=26907 comm="cp" name="build" dev="dm-0" ino=12651245 scontext=system_u:system_r:container_t:s0:c17,c902 tcontext=system_u:object_r:user_home_t:s0 tclass=dir permissive=0
type=ANOM_PROMISCUOUS msg=audit(1502369722.855:749): dev=veth0198420 prom=0 old_prom=256 auid=4294967295 uid=0 gid=0 ses=4294967295
```

Maybe something with SELinux?

hegjon, 2017-08-10:
> Try removing the directory first.
Same as before, and root owns the folder after the docker commands

prusnak, 2017-08-10:
Most, probably. If it works with `setenforce 0` I will have to close this as I do not know how to support SELinux

hegjon, 2017-08-10:
It workes after I disable SELinux with `setenforce 0`, I can spend some time figure out if we can solve it by adding a label to the build folder

hegjon, 2017-08-10:
This diff works on my setup with SELinux enforced:
```
diff --git a/build-firmware.sh b/build-firmware.sh
index 0603504..a9a9a32 100755
--- a/build-firmware.sh
+++ b/build-firmware.sh
@@ -6,7 +6,7 @@ TAG=${1:-master}
 BINFILE=build/trezor-$TAG.bin
 
 docker build -t $IMAGE .
-docker run -t -v $(pwd)/build:/build $IMAGE /bin/sh -c "\
+docker run -t -v $(pwd)/build:/build:z $IMAGE /bin/sh -c "\
    git clone https://github.com/trezor/trezor-mcu && \
    cd trezor-mcu && \
    git checkout $TAG && \
```
Docker will relabel the build folder as seen here:
```
$ ls -laZ build
total 388
drwxr-xr-x. 1 root  root  system_u:object_r:container_file_t:s0     34 Aug 10 15:29 .
drwxrwxr-x. 1 jonny jonny unconfined_u:object_r:user_home_t:s0     788 Aug 10 15:28 ..
-rwxr-xr-x. 1 root  root  system_u:object_r:container_file_t:s0 393412 Aug 10 15:29 trezor-v1.5.1.bin
```

hegjon, 2017-08-10:
The same issue is with build-bootloader.sh

hegjon, 2017-08-10:
More details here: http://www.projectatomic.io/blog/2016/03/dwalsh_selinux_containers/

prusnak, 2017-08-10:
Fixed in https://github.com/trezor/trezor-mcu/commit/f9af87091950ced7780df57e4a49e106d5369a7f

Thank you!

closed 2017-08-10 by prusnak

hegjon, 2017-08-10:
Great, you are welcome