trezor/trezor-mcu #211: Re-distribute the firmware in the Fedora repository?

hegjon, 2017-08-15:
Hi, I am wondering if re-distributing the firmware in the Fedora repository is something that you would advice against?

Advantages:
- We can run automated checks that verify that the firmware is correct, like we do for Electrum [1].

Disadvantages:
- Users and upstream (Trezor) need to trust a third-party (Fedora).

Fedora allows including the firmware as pre-built [2], the spec file and build is open for everyone to verify.

Creating this issue was the easiest way to have a public open discussion. I will not create a package if Trezor discourage it.

[1] https://src.fedoraproject.org/rpms/electrum/blob/master/f/electrum.spec#_49
[2] https://fedoraproject.org/wiki/Firmware

saleemrashid, 2017-08-15:
> Users and upstream (Trezor) need to trust a third-party (Fedora).

If Fedora distribute unsigned copies of the firmware then the security of the TREZOR is reduced significantly (because the user cannot necessarily tell if they are running Fedora's firmware or a malicious actor's firmware).

If you want to do this, you need to use the exact same ARM compiler and you should then get the same firmware as TREZOR distribute (you will be able to use the SatoshiLabs signatures). Otherwise, I recommend against it.

prusnak, 2017-08-15:
I think the idea was to distribute a signed firmware, basically as a binary blob.

I just don't understand how exactly would a Fedora user benefit from this packaged firmware.

hegjon, 2017-08-16:
The idea/steps were something like:

1. Use the unmodified firmware that is hosted on https://wallet.trezor.io/data/firmware (no building)
2. Run checks that verifies that the firmware is correct during the RPM package
3. The RPM package is built and signed on the same infrastructure as other Fedora packages

saleemrashid, 2017-08-16:
@hegjon How do you verify a binary blob without building it from source?

hegjon, 2017-08-16:
> @hegjon How do you verify a binary blob without building it from source?

There are different options, not sure if any of the options will be good enough:
- Someone from upstream sign the binary with their private GPG key, like we do for Electrum [1].
- Check that the fingerprint is the same as releases.json and/or build it with Docker and use these fingerprints for verification. 
- Building with the same tool-chain is an option, but I think it will be too much hassle to get the fingerprint correct.

I am sure there are other verification methods that I have not thought about.

[1] https://src.fedoraproject.org/rpms/electrum/blob/master/f/electrum.spec

hegjon, 2017-08-16:
> I just don't understand how exactly would a Fedora user benefit from this packaged firmware.

I have been going back and fourth, so it helps me a bit to just get the idea out of my head if there are no benefits or if it compromises the security. It might be that I lack some knowledge about the verification work-flow. Can an attacker replace release.json and firmwares with the malicious actor's firmware?

One use case:
Advance users can download the firmware via RPM packages and run `tail -c +257 trezor-firmware.bin | sha256sum` and compare that with release.json hosted wallet.trezor.io.

saleemrashid, 2017-08-16:
@hegjon That doesn't achieve much because the TREZOR does that anyway. Building it from source, however, provides assurance that SatoshiLabs is not a bad actor.

saleemrashid, 2017-08-16:
> It might be that I lack some knowledge about the verification work-flow. 

The firmware is signed by SatoshiLabs and the bootloader verifies this. A malicious actor, therefore, would not be able to replace the firmware without the user knowing that it was not official firmware.

However, this doesn't protect you if SatoshiLabs is the malicious actor but you can verify this by building the firmware yourself and checking if the hash checks out.

prusnak, 2017-10-27:
Closing. If there is anything we can do, feel free to ask or reopen.

closed 2017-10-27 by prusnak